EdgeTX/edgetx #508: New Bluetooth Protocoll

brainbubblersbest, 2021-07-27:
On opentx is an old pull request with work from @bsongis
https://github.com/opentx/opentx/pull/7198#issue-353440589

I dont know what are the Main conflicts that need more additional Work. But this Code should not be Forgotten.

If this Code has the possible functions to configure generic BT Module Hardware, like HM-10 or hc-05 over AT Commands, then this can be a solution to get rid from proprietary FRSKY Hardware too.

I dont think this is a urgent Feature. But i think it should be linked here to remember on this unfinal feature.

raphaelcoeffic, 2021-07-28:
> On opentx is an old pull request with work from @bsongis
> [opentx/opentx#7198 (comment)](https://github.com/opentx/opentx/pull/7198#issue-353440589)
> 
> I dont know what are the Main conflicts that need more additional Work. But this Code should not be Forgotten.
> 
> If this Code has the possible functions to configure generic BT Module Hardware, like HM-10 or hc-05 over AT Commands, then this can be a solution to get rid from proprietary FRSKY Hardware too.
> 
> I dont think this is a urgent Feature. But i think it should be linked here to remember on this unfinal feature.

You mean like Mike’s ErSkyTX? I think he supports all that.

brainbubblersbest, 2021-07-28:
I never tried erskytx to be honest.
But the actual. BT Integration feels let me say... 
as it is in sort of an alpha state. 

I think no one still expect further Information from Frsky or a Firmware fix for These Modules in the Future. 😅
The Code from bsongis i think is more target to the Frsky Para Modules? 

When the Code for generic Modules exist already in ersky, then it Sounds not Bad at all to merge it into edge? 🙃
What Do you say?

raphaelcoeffic, 2021-07-28:
> I never tried erskytx to be honest.
> But the actual. BT Integration feels let me say...
> as it is in sort of an alpha state.
>
> I think no one still expect further Information from Frsky or a Firmware [...]

ErSkyTX is quite different, I believe there is some real work involved 😬