EdgeTX/edgetx #509: Sbus Trainer over Serial RX in JR Module Bay/LiteBay on Externalaccess capable Radios

brainbubblersbest, 2021-07-27:
This should be a nice feature to receive originally inverted sbus Protocol (R-XSR or other receivers and external mpm) 
Over the inverted highspeed serial RX Port on lite bay and JR Bay external Access capable radios. 
For Trainer/slave Usage. And without the need of a 3,5mm plug/cable or everything else. 

External Multimodules Supporting this Feature too, as i understand it correctly. 
https://github.com/opentx/opentx/pull/5518#

raphaelcoeffic, 2021-07-28:
> This should be a nice feature to receive originally inverted sbus Protocol (R-XSR or other receivers and external mpm)
> Over the inverted highspeed serial RX Port on lite bay and external Access (Mod) capable JR Bay Radios.
> For Trainer/slave Usage. And without the need of a 3,5mm plug/cable or everything else.
> 
> External Multimodules Supporting this Feature too, as i understand it correct.
> [opentx/opentx#5518](https://github.com/opentx/opentx/pull/5518)
> Please read to the End.
> @raphaelcoeffic know it for shure 😉
> 
> I would really like to use it with my x12s and the x-lite for Rock solid wireless Trainer Sessions.
> 
> It's the oldest non solved Pull request on Opentx.
> So it seems to be like a Thing that feels good to make it finally possible 😁 😋

I can tell you that the wireless trainer based on MPM works really well! I’ve use it countless times.

brainbubblersbest, 2021-07-28:
Over s.port?

raphaelcoeffic, 2021-07-28:
> Over s.port?

sure, the MPM sends telemetry back over S.Port. If you have an internal ISRM, this is no problem, just need to cut one useless line, and you're done.