EdgeTX/edgetx #511: eachine t16s

pds59, 2021-07-28:
can not upgrade internal 4-1 module error: firmware update error no sync

pds59, 2021-07-28:
firmware : multi-stm-serial-aetr- v1.3.2.86

rotorman, 2021-07-28:
Which EdgeTX firmware are you using (GitHub hash number in System -> Version screen)?

pfeerick, 2021-07-28:
Also, is the module enabled? i.e. when you go into the model settings for
the currently active model, internal RF is set to MULTI, and you can see
the MPM status and version.

pds59, 2021-07-29:
Version 2.5.0 (46688263

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

pds59, 2021-07-29:
Module enable,  internal rf set to multi,  now afterI did try to upgrade the internal module module status is as follow: no multi telemetry detected 

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

pds59, 2021-07-29:
Now all my active model with the same status : the internal module module status is as follow: no multi telemetry detected

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

MervDale, 2021-07-29:
Just  a note I have the Eachine TX16S and I uploaded the latest firmware for the 4 in 1 Module from the firmware folder on the SD card with no issues so if it is not working on your radio you may have a hardware problem

pds59, 2021-07-29:
I did the same but di not workPaolo de scaglia

rotorman, 2021-07-31:
We have a further report on Discord with RadioKing TX18S having the same issue:
https://discord.com/channels/839849772864503828/839895574244229152/870297350084636754
Seems to be issue only with ETX, OTX apparently works:
https://discord.com/channels/839849772864503828/839895574244229152/870857929002209350

pds59, 2021-07-31:
For your reference i was force ti reset the radio and reload all (sistem, sd card etc)Paolo de scaglia

https://discord.com/channels/839849772864503828/839895574244229152/870297350084636754
https://discord.com/channels/839849772864503828/839895574244229152/870857929002209350

rotorman, 2021-07-31:
MervDale reported on Discord that RadioKing TX18S was able to update MPM using Edge 2.4 without issues:
https://discord.com/channels/839849772864503828/839895574244229152/870973940288323584
I wonder if it is then an issue with 2.5 or an issue with @pds59 radio. We need another TX18S user test this with 2.5 and report I guess.

rotorman, 2021-07-31:
Just tested flashing 4-in-1 on RM TX16S to MPM fw v1.3.2.86 (from v1.3.2.77) under EdgeTX 2.5.0 (with b7d95b2e), no issues.

pfeerick, 2021-08-01:
I just flashed my RM TX16S's internal MPM to v1.3.2.86 AETR back to v1.3.2.86 TAER (i.e. changed the default channel ordering) with a build from from last week (55098b66) and it updated fine. Not looking promising as a reproducible issue unless it is TX18S specific. 😢

pfeerick, 2021-09-02:
@pds59 Have you tried this again since or is this still an issue? Have you been able to identify this as an EdgeTX issue, or does OpenTX suffer the same problem on this radio (thus mostly eliminating a hardware issue).

pds59, 2021-09-02:
Hi, i have a radiomaster t16s with open tx and eachine t16s with edge tx (basically same radio) after I full reload edge tx I was able to upgrade external module 

tks

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

rotorman, 2021-09-02:
Thx for the feedback. Closing the issue.

closed 2021-09-02 by rotorman