EdgeTX/edgetx #515: Time settings are lost after restarting TX

eshifri, 2021-07-29:
**Describe the bug**
Time settings are lost after restarting TX

**To Reproduce**
Set current date and time. The settings are correctly displayed in the top bar.
Switch TX OFF and ON again - date/time are set to 0.

**General information**
 - EdgeTX version current code
 - Hardware NV14

raphaelcoeffic, 2021-07-30:
I noticed that as well, but then I measured the RTC battery, and it has 2.3V in my case. I asked for information on this, as it soldered onto the main board with a carrier welded to the battery.

rotorman, 2021-07-30:
I checked the VBAT voltage data at [STM32F429BI datasheet](https://www.st.com/resource/en/datasheet/stm32f429bi.pdf).
Section 6.3.1 on page 95 lists VBAT voltage to be between 1.65 and 3.6V. The same is also listed under Section 3.16 on page 26 and also the Figures 25 and 26 about VBAT current consumption list 1.65V, thus I am pretty sure, 2.3V should be OK and the error lies somewhere else.

rotorman, 2021-07-30:
According to this post it should be 3V battery in NV14:
https://www.rcgroups.com/forums/showthread.php?3023832-UndergroundFPV-Dark-Knight-Radio/page163#post42146103

raphaelcoeffic, 2021-07-30:
> According to this post it should be 3V battery in NV14:
> https://www.rcgroups.com/forums/showthread.php?3023832-UndergroundFPV-Dark-Knight-Radio/page163#post42146103

Ok, thx for the Research work! I guess the RTC is probably just not initialized and/or not read on start.

rotorman, 2021-07-30:
You are likely spot on about the RTC init. Whereas horus initializes rtc:
https://github.com/EdgeTX/edgetx/blob/46688263eec762fe2359612e3db1677a0e08cad3/radio/src/targets/horus/board.cpp#L242
there is no rtcinit ever called for NV14:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/targets/nv14/board.cpp

eshifri, 2021-07-30:
Original NV14 file does not have rtcInit() either:
https://github.com/eshifri/Flysky-OpenTX-Test-Branch/blob/chevy/NV14-flysky/radio/src/targets/nv14/board.cpp
but time is preserved there.

raphaelcoeffic, 2021-07-30:
> Original NV14 file does not have rtcInit() either:
> https://github.com/eshifri/Flysky-OpenTX-Test-Branch/blob/chevy/NV14-flysky/radio/src/targets/nv14/board.cpp
> but time is preserved there.

I guess the RTC must be initialised somewhere, otherwise it won't work.

eshifri, 2021-07-30:
I would think so. :-) I will search the code later today.

rotorman, 2021-07-30:
The code you linked initializes RTC here:
https://github.com/eshifri/Flysky-OpenTX-Test-Branch/blob/3ba03c385ba3107d5c012eb2f54da97b5d2439fb/radio/src/opentx.cpp#L2417-L2419

eshifri, 2021-07-30:
Adding rtcInit() to the board.cpp fixies the problem.

rotorman, 2021-07-30:
@eshifri Would you do a PR with the fix please?

closed 2021-08-03 by raphaelcoeffic