EdgeTX/edgetx #516: NV 14 openTX models are not converted when ported to edgeTX.

eshifri, 2021-07-29:
See title.
**General information**
 - EdgeTX version current code
 -  - Hardware NV14

raphaelcoeffic, 2021-08-01:
@eshifri can you please provide me with a most complete model together with what is configured so I can try it out? Normally I would use companion for that, but the NV14 is not supported in companion.

eshifri, 2021-08-01:
[model14.zip](https://github.com/EdgeTX/edgetx/files/6913257/model14.zip)
This is a single motor plane, Rudder, elevator, flapperons; three rates; some mixies; FMs are not used; telemetry: RSSI and battery voltage; logical switches. RF power set to standard (in NV14 OTX there is an option to set high power). RSSI output set to "native" - also special feature - not really important. These two NV14 special things are encoded in bit-fields, that used to be "reserved".

nightflasher, 2021-09-21:
Still the case on 2.5.0 RC1 - gets more problematic, as the edgeTX wipes all models.

closed 2021-12-04 by eshifri