EdgeTX/edgetx #519: NV14 bootloader crashes on exit

eshifri, 2021-07-29:
**Describe the bug**
When exiting NV14 bootloader the screen is covered with vertical colored lines and radio switches off.

**To Reproduce**
Enter bootloader either with left trim down or selecting EXIT with right trim the TX screen goes into typical crash color pattern and the radio switches off.

Not a big deal, since all operations can be performed, but...

**Expected behavior**
Similar to RM TX16 go to TX boot.

**General information**
 - EdgeTX current.
 - Hardware NV14

closed 2021-11-01 by raphaelcoeffic

Syncmasterklop, 2021-11-27:
The problem was not solved.

pfeerick, 2021-11-27:
Sorry? If you are running a nightly bootloader from within the last 26 days then it is correctly exiting to the firmware for me. If you are running at 2.5.0 bootloader then it will still be broken for you.

Syncmasterklop, 2021-11-27:
I have updated the last nightly build, the problem remains. Am I doing something wrong?
https://1drv.ms/v/s!AlG8PMfhz1cTqRxJ-sDLEFnpwX_i

raphaelcoeffic, 2021-11-27:
> I have updated the last nightly build, the problem remains. Am I doing something wrong?
> https://1drv.ms/v/s!AlG8PMfhz1cTqRxJ-sDLEFnpwX_i

This is showing the firmware installed, not the bootloader. The question is: did you update with DFU mode, or not?

Syncmasterklop, 2021-11-27:
No, I "updated" the bootloader through the file manager with the command "flash bootloader". Now I will try through DFU.

raphaelcoeffic, 2021-11-27:
> No, I "updated" the bootloader through the file manager with the command "flash bootloader". Now I will try through DFU.

Well, it should be equivalent. But what do you consider “crashing”. When you hit exit: does it reboot or not?

Syncmasterklop, 2021-11-27:
I showed it in the video, and in the first message there is an exact description "When exiting NV14 bootloader the screen is covered with vertical colored lines". This does not bother me much, I just wanted to point out that the problem is not fully resolved.

raphaelcoeffic, 2021-11-27:
> I showed it in the video, and in the first message there is an exact description "When exiting NV14 bootloader the screen is covered with vertical colored lines". This does not bother me much, I just wanted to point out that the problem is not fully resolved.

Well, this is not "the crash", it's just the screen not being cleared before reboot ;-) Before it was fixed, it would not reboot at all and needed to be powered ON again after exiting.

commit 766ee208 (2021-10-31): Fix some power-up issues

Resolves #519
commit 3247b59f (2021-11-01): Fix some power-up issues

Resolves #519