EdgeTX/edgetx #520: NV14 internal module does not have any protocols

eshifri, 2021-07-29:
**Describe the bug**
When setting up a new model the only option available for internal module is OFF.
External module has lots of options, although it is not  present. For any external option protocol selection is allowed. 

**To Reproduce**
Create a new model, go to modules.

**Expected behavior**
Internal module should support AFHDS2A protocol.
External option should not be displayed if HW is not present. It is not the case on other radios, so, probably, is OK.

**General information**
 - EdgeTX current code.
 - Hardware NV14

eshifri, 2021-07-30:
It was partially my mistake: I did not realize that NV14 version has to be compiled with -DAFDHS3=YES.
After doing this I got the FlySky internal module, but selecting AFHDS2A as subprotocol caused a crash.

raphaelcoeffic, 2021-07-30:
> It was partially my mistake: I did not realize that NV14 version has to be compiled with -DAFDHS3=YES.
> After doing this I got the FlySky internal module, but selecting AFHDS2A as subprotocol caused a crash.

Which in this case is a synonym for „not yet implemented” 😅

eshifri, 2021-07-30:
Hmmm. In the NV14 repository SkyHly module was defined as a separate option.
In edgeTX this module type does not exist and isFlySkyModule() compares current module with AFHDS3. AFHDS2A is implemented (or defined) as subprotocol.
Are you saying that AFHDS2A support is not implemented yet?

BTW, my TX went into permanent EMERGENCY MODE. No way to take it out of it. Even after going back to OTX 2.1 and returning to edgeTX it it stays in emergency mode.
Any suggestions?

rotorman, 2021-07-30:
UNEXPECTED_SHUTDOWN=NO CMake?

eshifri, 2021-07-30:
Thank you! It helps.

raphaelcoeffic, 2021-07-31:
> Hmmm. In the NV14 repository SkyHly module was defined as a separate option.
> In edgeTX this module type does not exist and isFlySkyModule() compares current module with AFHDS3. AFHDS2A is [...]

Ok, so I should fix this first thing! This could confuse quite some people!

eshifri, 2021-08-03:
Just some considerations...
AFHDS3 is supported with eternal module only, so it cannot be an option for internal one - will be too confusing even if it works.
AFHDS2A cannot be a subprotocol of AFHDS3.
I remember seeing a comment somewhere in the code saying that also slots for module names are already used and FlySky module is a new module type for OTX/edgeTX.

raphaelcoeffic, 2021-08-03:
> Just some considerations...
> AFHDS3 is supported with eternal module only, so it cannot be an option for internal one - will be too confusing even if it works.

Agreed! This was just my try at "fixing" the fact we don't have any module IDs left to use, So I had to use a sub-type. The module type is then "FlySky", with "AFHDS3" and "AFHDS2A".

> AFHDS2A cannot be a subprotocol of AFHDS3.
> I remember seeing a comment somewhere in the code saying that also slots for module names are already used and FlySky module is a new module type for OTX/edgeTX.

I changed the module type label from "AFHDS3" to "FlySky" and added the 2 sub-types. And yes, the change is not 100%, that was just the first try.

richardclli, 2021-08-05:
The latest code for AFHDS2A is in this repository, OpenTX 2.1.18+ of current stable NV14 firmware.
https://github.com/open-flysky/Flysky-OpenTX-Test-Branch

richardclli, 2021-08-21:
> 
> 
> Just some considerations...
> AFHDS3 is supported with eхternal module only, so it cannot be an option for internal one - will be too confusing even if it works.
> AFHDS2A cannot be a subprotocol of AFHDS3.
> I remember seeing a comment somewhere in the code saying that also slots for module names are already used and FlySky module is a new module type for OTX/edgeTX.

Not necessary true, as I know Flysky may have internal AFHDS3 module for NV14 in the coming future.

eshifri, 2021-08-21:
OK, but wouldn't it be a different module?
For NV14 with internal module only, AFHDS3 should not be available in the protocol selection, right?
Anyway, the main problem at this point, IMHO is we ran out out available bits for modules.
If we use FlySky for module and AFHDS3/AFHDS2A as subprotocols we need to use additional field (mode in NV14 OTX) for PWM/SBUS, PWM/iBus, etc.
And still we need to find the way to disable AFHDS3 as internal protocol for NV14. But as internal only - external is OK.
And if some new module needs to be supported, we are back to square one.

pfeerick, 2021-08-21:
I think it just means if there is something on the horizon, any solution done now should be mindful of that where possible, nothing more. As things stand right now, AFHDS v3 is external only (as the PL18 isn't a supported... yet?!)... and is it's own protocol. If an upgraded NV14 is anything like the PL18, it will be AFHDS3 only, so that may not be such an issue - it can be a NV14+ or something that has AFHDS3 instead of AFHDS3 as the internal RF. 

Since the constraint there is 4 bit value (`enum ModuleType`) ... if doubling the bits for that to 8 bit is problematic, how about filtering it a bit better - i.e. I see five types of R9, two XJTs, ... can they be broken out to a second list?

eshifri, 2021-08-21:
I do not know whether this structure is persisted in the model.
If it is, such change will complicate conversion significantly.

richardclli, 2021-08-23:
So the 4 bits limit is for persisting the settings, well, why not change it once and for all?  When we supports more radios, there are possibilities of more internal module types.

eshifri, 2021-08-23:
Makes sense to me, but then we have to absorb the pain of models conversion now.

raphaelcoeffic, 2021-08-23:
> So the 4 bits limit is for persisting the settings, well, why not change it once and for all? When we supports more radios, there are possibilities of more internal module types.

The idea was to cope with it right now with the sub-types, and have the module types extended once YAML is in place, as it simplifies conversions.

eshifri, 2021-08-23:
> The idea was to cope with it right now with the sub-types, and have the module types extended once YAML is in place, as it simplifies conversions.

That was my understanding as well, but the problem is AFHDS3 and AFHDS2A are physically different modules and they both may be present on the same NV14; one - internal, the other - external. So, maybe, this is the way to handle it now.
So it is not enough to have a FlySky type, since we cannot allow both sub-protocols on the same module.
If in the future such module may be available, the entanglement may be unresolvable.
Changing this structure may have a domino effect no matter what we do.

raphaelcoeffic, 2021-08-23:
> That was my understanding as well, but the problem is AFHDS3 and AFHDS2A are physically different modules and they both may be present on the same NV14; one - internal, the other - external. So, [...]

This not necessary: if you have both at the same time, one internal, and one external, you would have the following:
- internal: Flysky / AFHDS2
- external: Flysky / AFHDS3

You can think of the subtypes just as an extension of the module type. All it does it to require a second drop-down to allow for it to be chosen. As the internal module is fixed, both drop-downs would be "locked", or we would even only display the second, that's not an issue.

> If in the future such module may be available, the entanglement may be unresolvable.
> Changing this structure may have a domino effect no matter what we do.

Once we can flexibly change the bit-mapping, this can be modified easily.

iggut, 2021-09-01:
Any chance we could please get a complied version that has AFHDS2 enabled for internal module?

richardclli, 2021-09-02:
The problem is EdgeTX source code do not have AFHDS2 protocol driver, need to port either from Open-Flysky or OpenTX (Jakub's branch) repository.

eshifri, 2021-09-13:
I do not think it is that big of a problem to port Jakub's code, but I do not see consensus how to handle this extra module type.

raphaelcoeffic, 2021-09-13:
> I do not think it is that big of a problem to port Jakub's code, but I do not see consensus how to handle this extra module type.

That is for me the smaller issue. All that’s needed is a clear pointer at which version is the latest. Then I can probably take it from there.

eshifri, 2021-09-14:
Then we are in luck!
This is the latest tested version: https://github.com/open-flysky/Flysky-OpenTX-Test-Branch

richardclli, 2021-09-14:
> 
> 
> Then we are in luck!
> This is the latest tested version: https://github.com/open-flysky/Flysky-OpenTX-Test-Branch

This is OpenTX 2.1.x branch, and Jakub have another target for OpenTX 2.4, in this branch:

https://github.com/opentx/opentx/tree/qba667/nv14merge

But, this version have a lot of changes in pulses underlying layer, in which I tried a few times and still cannot figure out how to merge and make it works.

closed 2021-09-21 by raphaelcoeffic

eshifri, 2021-09-30:
> This is OpenTX 2.1.x branch, and Jakub have another target for OpenTX 2.4, in this branch:
>
> https://github.com/opentx/opentx/tree/qba667/nv14merge
>
> But, this version have a lot of changes in pulses [...]

Actually the https://github.com/open-flysky/Flysky-OpenTX-Test-Branch has last commit from June 23 2021, while 
https://github.com/opentx/opentx/tree/qba667/nv14merge - March 2020.

I believe there were some important fixies after March 2020.

eshifri, 2021-09-30:
E.g. telemetry_nv14.cpp was seriously modified.