EdgeTX/edgetx #521: Only 12 channels in the MODEL SETUP.

Salto42, 2021-07-29:
In the MODEL SETUP with the Frsky D16 protocol we have access to only 12 channels and the same goes for the FAILESAFE SETINGS in Custom, Set.
But everything is fine for the MIXES.
( T16 )

pfeerick, 2021-07-31:
It seems that 'old' models with 16 channels show up as 1 - 16, but 'new' models max out at 12. Even so, an 'old' model that listed 16 channels only showed 12 in the failsafe -> custom -> set screen. And any changes to the field automatically limits it to 12 channels again.

pfeerick, 2021-07-31:
Basically same as was reported in #299.

raphaelcoeffic, 2021-08-03:
Ok, so we need some basic recipe to tell how many max channels there are depending on module/protocol. This is absolutely not clear to me and should be researched. Once this info is secured, let’s re-implement the method to get that maximum, as it clearly does not work now.

richardclli, 2021-08-17:
Any logic can be digged out in OpenTX 2.3 branch?

closed 2021-08-19 by pfeerick