EdgeTX/edgetx #522: Feature request - Additional higher baud rates

JyeSmith, 2021-07-29:
Most handset currently have 115200 and 400k but are capable of much higher bauds.  3750000 has been tested on an x7 with the Rx inverter mod and appears to be running without issue.  It would be great to have a number of higher options available to users for further testing.

Additional changes are also needed to increase the GPIO speed and change oversampling to 8.

`GPIO_InitStructure.GPIO_Speed = GPIO_Speed_25MHz;`

`USART_OverSampling8Cmd(TELEMETRY_USART, ENABLE);`

```
  3750000,
  1870000,
  1000000,
  400000,
  115200,
```

raphaelcoeffic, 2021-08-04:
Do you think we should have the GPIO speed & oversampling setting set for all bitrate by default? Or is this something that we should turn on from a certain speed on?

JyeSmith, 2021-08-04:
I think so.  Ive put a number of packs through the x7 on 3.75Mbps with `USART_OverSampling8Cmd` with out issue.  Am I correct thinking it should be less of an issue at lower baud?

Below is what I currently have flashed at testing on various modules.

```
115200
400000
921600
1870000
3750000
```

raphaelcoeffic, 2021-08-04:
@JyeSmith We still need to find a solution for how the module & ETX agree on which baudrate to use. I think we will make a couple of sub-types for CRSF: `TBS` & `ELRS`. That should give us the possibility to have different negotiation methods / set of bitrates, thus giving you the full freedom.

JyeSmith, 2021-08-04:
I think for now manually selection is fine until we get some feedback.  Users are already comfortable switching between 115200 and 400k.

closed 2021-09-11 by pfeerick