EdgeTX/edgetx #524: NV14 cannot exit EMERGENCY MODE

eshifri, 2021-07-30:
**Describe the bug**
If NV14 SW crashes and the TX goes into emergency mode there is no way to return it to normal mode.
Switching it off (remove batteries) does not help. Re-flashing the FW - either.

**General information**
 - EdgeTX current code
 - Hardware NV14

raphaelcoeffic, 2021-07-30:
> **Describe the bug**
> If NV14 SW crashes and the TX goes into emergency mode there is no way to return it to normal mode.
> Switching it off (remove batteries) does not help. Re-flashing the FW - either.
> 
> **General information**
> 
> * EdgeTX current code
> * Hardware NV14

This possibly related to the RTC, as it uses it’s ram to store some info on reboot/boot reasons.

rotorman, 2021-07-30:
@eshifri before the fix to this issue is found - in order for you to be able to use NV14 - try building with UNEXPECTED_SHUTDOWN=NO and see if this at least allows you to boot the NV14 again normally. UNEXPECTED_SHUTDOWN=NO basically should remove the EM screen.

eshifri, 2021-07-31:
Both OTX and edgeTX in boardInit() have the following comment:
rtcInit(); // RTC must be initialized before rambackupRestore() is called
but opentx.cpp calls rambackupRestore() prior to boardInit().
For some reason ot does not create problems for RM TX16S, but for NV14 - does.

eshifri, 2021-08-03:
Fixed with #542

closed 2021-08-03 by eshifri