EdgeTX/edgetx #526: ColorEdit() crashes on NV14

eshifri, 2021-07-31:
**Describe the bug**
ColorEdit() crashes on NV14 (at least in the Widgets setup)

**To Reproduce**
Go to widgets setup (top bar or main screen). Select any widget that allows to change color. E.g. Value or Text. Touch any of the color selection boxes. TX crashes prior to bringing virtual keyboard.

**General information**
 - EdgeTX version  current code.
 - Hardware NV14

eshifri, 2021-08-03:
One more peace of information: the problem is not in the ColorEdit() per. se. Setting the widget background works fine. The crash happens only when trying to change font color.

eshifri, 2021-08-04:
I think the problem is NumberEdit() called from ColorEdit() is trying to open the virtual keyboard inside the widget.

raphaelcoeffic, 2021-08-04:
> One more peace of information: the problem is not in the ColorEdit() per. se. Setting the widget background works fine. The crash happens only when trying to change font color.

@pfeerick found some weird stuff regarding widgets & color settings (see #544). This might be related as well. Getting this to crash on the simulation with stack trace would help greatly fixing this!

eshifri, 2021-08-04:
I'm not sure it is related: on NV14 the crash occurs when the system tries to create a virtual keyboard. On TX16S with the rotary encoder there is no problem at all.

raphaelcoeffic, 2021-08-04:
> I'm not sure it is related: on NV14 the crash occurs when the system tries to create a virtual keyboard. On TX16S with the rotary encoder there is no problem at all.

From I could see while debugging, the reason for this is actually the "Widget Settings" window and the method with which the virtual keyboard looks for a suitable window to hook itself into:
https://github.com/EdgeTX/libopenui/blob/c350cbfc9a43f42cc1edc5ae68f2b8d03cb5c613/src/keyboard_base.h#L76-L86

And here as well:
https://github.com/EdgeTX/libopenui/blob/c350cbfc9a43f42cc1edc5ae68f2b8d03cb5c613/src/keyboard_base.h#L55-L60

I can prevent the `nullptr` access, but that does not solve the issue: the virtual keyboard is looking for an `OPAQUE` window to push it "up" and insert itself in the room made bellow. As the widget settings is a pop-up window, that strategy does not work, so that a better method is need which would adapt to this kind of situations.

raphaelcoeffic, 2021-08-04:
@eshifri I might have a simple solution: EdgeTX/libopenui#11

eshifri, 2021-08-04:
Looks promising!
BTW, I have just realized that Text widget has exactly the same problem (crash).

eshifri, 2021-08-04:
So the virtual keyboard will always be in the main window?
I believe the same was done in the OTX NV14.

raphaelcoeffic, 2021-08-05:
> So the virtual keyboard will always be in the main window?
> I believe the same was done in the OTX NV14.

Not exactly: it’s “pushing” the closest full screen window aside to which the field to be edited belongs. This is to avoid sub-forms inside a page.

eshifri, 2021-08-05:
I see. Thank you for the explanation.
BTW, something is wrong with Windows builds: no matter which target I select it builds simulator and fails with 

CMake Error at radio/src/targets/simu/CMakeLists.txt:74 (target_link_libraries):
  Cannot specify link libraries for target "PRIVATE" which is not built by
  this project.

raphaelcoeffic, 2021-08-05:
> I see. Thank you for the explanation.
> BTW, something is wrong with Windows builds: no matter which target I select it builds simulator and fails with
>
> CMake Error at [...]

@rotorman noticed this yesterday. I just posted a "hotfix" (5cd68753d37285e7). However, the Companion-build still breaks, and it is not yet clear why. When requiring the higher CMake version it seems to fail to find some Qt libs (`QNetworkAccessManager`, `QPrinter` & `QtPrintSupport`). It is not clear yet why.

raphaelcoeffic, 2021-08-05:
> Looks promising!
> BTW, I have just realized that Text widget has exactly the same problem (crash).

Can you please check that the fix I provided does indeed fix it as well? It would be cool to check whether it has any other ill effects or not.

eshifri, 2021-08-05:
I was not able to try it.
It is weird, but after the rebase I can compile the FW, but it does not boot.
I'm getting a blink on the screen and it hangs prior to showing the logo.
In order to get to the bootloader screen I have to take out the battery.
I have tried several clean builds - does not help.

rotorman, 2021-08-05:
@eshifri you might like a debugger with Serial Wire Debug hooked up to a radio - then you can step through the code from your PC/Mac and see where it hangs, inspect the variables and so on. I am very fond of [Segger Ozone](https://www.segger.com/products/development-tools/ozone-j-link-debugger/) for debugging, works beautifully with EdgeTX as well.

eshifri, 2021-08-10:
@raphaelcoeffic 

> Can you please check that the fix I provided does indeed fix it as well? It would be cool to check whether it has any other ill effects or not.

Tested it. Works nice. I have not found any side effects.

closed 2021-08-18 by raphaelcoeffic

pfeerick, 2021-08-19:
This is either back again, or wasn't actually fixed by that PR... 478e345a is crashing for me exactly following the steps @eshifri described.

eshifri, 2021-08-19:
478e345 is for bootloader.
ColorEdit work for me with this change (built the FW yesterday).

eshifri, 2021-08-19:
Oops, I have just realized that I do not have yesterday's changes for windows.h

pfeerick, 2021-08-19:
> [478e345](https://github.com/EdgeTX/edgetx/commit/478e345a32ab299402bf891096b3b6b2957af862) is for bootloader.

Yes, and no. It's actually yesterdays nightly also, which includes EdgeTX/libopenui/pull/11 :-P 

I just thought there might have been some confusion as there were two conversations going on - initially the bug, then talk about the simulator - and wasn't sure which exactly you meant was working since it could have been that the "hotfix" worked.

eshifri, 2021-08-20:
I meant ColorEdit is working.
Have not tried the simulator.

pfeerick, 2021-08-20:
Can you try the nightly (i.e. have to download it before it gets rebuilt in about two hours) when you get a chance so we're comparing exactly the same builds - as touching the edit fields for either text or value widgets colour settings crashes for me.

eshifri, 2021-08-20:
Sorry, will not have time for this in next 2 hours.

pfeerick, 2021-08-20:
Latest nightly is fine, so I don't know what's going on there... but it works.