EdgeTX/edgetx #535: [X12S] Add calibration for analog joystick

raphaelcoeffic, 2021-08-01:
The title says it all :-)

Luznatural, 2022-11-18:
Please, do not forget about this issue.
Thanks in advance

1Sven, 2022-11-18:
Yes, please do it. 😉

Luznatural, 2022-12-04:
Any progress on this issue?
Thanks in advance

Luznatural, 2023-02-03:
I hope that we have taken in account that when the calibration of the analog joystick is fixed, it will also means that you could use that analog stick in the X12S as an input for producing outputs you want to perform, with mixes in the middle, like any other joystick.
Thanks developers, you are doing a great job for saving X12S users from oblivion.

pfeerick, 2023-04-13:
No, as there is no implementation for this yet. This is an issue so that it doesn't get forgotten.

raphaelcoeffic, 2023-09-05:
Just to be clear about the actual status of this: it is actually only the visualisation that is missing, that joystick is actually calibrated if you move it properly during calibration.

Luznatural, 2023-09-05:
> Just to be clear about the actual status of this: it is actually only the visualisation that is missing, that joystick is actually calibrated if you move it properly during calibration.

Raphaelcoeffic, the problem is not only the calibration screen of the analog joystick (AJ). The problem is that EdgeTX doesn't recognize the AJ in any way.

The analog joystick was working in OpenTX as a totally functional third stick, with two differentiated movements: horizontal and vertical. When you release it, always returns to the center in both directions, like the right stick in any standar radio with mode 2 selected.

I have re-checked again with EdgeTX v2.9.0, and found the following:

- AJ is not listed in any of the pages nor options in the System/Hardware list.
- You cannot select the AJ as an input or source to produce any mix or output, when it should be treated like the other two standard joysticks.
- AJ is totally useless as it is now.

EdgeTX should be able to allow you to use the AJ to produce outputs. As an example, in the early versions of OpenTX (v 1.5.2) I was able to use the AJ as input for moving the gimbal of my FPV camera installed into the canopy of my planes. I moved it to look up, down, right or left progresivelly, like if I were onboard my plane.

I hope the EdgeTX team should hace mercy for us and look for a solution!!!

Frsky X12S is an oustanding (and expensive) radio that doesn't deserve yet to be retired of service.

You should be able to assign the analog joystick to perform any

raphaelcoeffic, 2023-09-05:
I don't think this is the case, you should see 2 analog sources:
- `JSx`
- `JSy`

Which version are you on? May I see your `radio.yml`?

Luznatural, 2023-09-05:
I am unable to see any of those.
Where do you have to look for finding them?
If I go to hardware and down to inputs, and I see calibratio, sticks, pots, sliders and switches. I do not find JSx JSy in any of the options.
I only see any analog indication in debug, and thereIf I move the analog joystick the numbers do not change.
Here you have my radio.yml file:
[radio.zip](https://github.com/EdgeTX/edgetx/files/12526767/radio.zip)
Thanks in advance

raphaelcoeffic, 2023-09-05:
This is `2.9.0` (no setting necessary in "Hardware"):

[image]

And it seems to be calibrated (not sure this is correct, but it looks like it):
```yaml
calib:
   MOUSE1:
      mid: 1019
      spanNeg: 697
      spanPos: 773
   MOUSE2:
      mid: 1007
      spanNeg: 817
      spanPos: 781
```

Luznatural, 2023-09-05:
Rafael,
Tonight I am going to sleep far better, having learned how to recover a past capability for my X12S.
You were totally right. I was able to assign both AJ's to the camera. I had to calibrate two times to get the AJ's reach -100 to +100, but I guess it was my fingers and not the software.
So it's just now a minor matter to include the graphics for AJ in the calibration screen.
Thank you very much
Best regards

grafwiloff, 2023-12-27:
💥🏁👍🏻ThanXs a lot, you are great!!!!
it works, you are doing a perfekt job!
happy new year…Matthias