EdgeTX/edgetx #536: Connecting USB (SD-Card access) crashes running EdgeTx

wimalopaan, 2021-08-01:
Title says it all

raphaelcoeffic, 2021-08-01:
> Title says it all

We will need a little more details! As of now, I can connect without issues on all my radios. So…. Which radio is that?

wimalopaan, 2021-08-01:
Oh sorry: it is a TX16s

rotorman, 2021-08-01:
Just tested RM TX16S with 09a65807 and Windows 10. I can attach USB, select "USB Storage (SD)", the two drives pop up on PC, and radio displays full screen USB logo.
I can disconnect from PC, the radio returns to EdgeTX user interface and still works fine.
Tested twice, no issues.

@wimalopaan what version of EdgeTX are you using?

wimalopaan, 2021-08-01:
I used 6e37d117 with Arch/Linux

pfeerick, 2021-08-01:
I was running a derivative of Arch until about a week ago, and didn't have any issues with my TX16S. When you say it crashes, what exactly happens? Just freezes? Usb banner not shown? Screen goes blank? Emergency mode? 😂

wimalopaan, 2021-08-02:
It shows the popup to choose the mode. If you select SD-Card, it shows the USB logo and then crashes. It just turned completely off without any further message. Blank screen and led goes off.
When the powring up again it shows emergency mode.
After power cycling again it reboots normally.

3DRacer, 2021-08-02:
try please. secure sd content on pc. format the sd card. play files back again. i had the same problem before

wimalopaan, 2021-08-02:
> try please. secure sd content on pc. format the sd card. play files back again. i had the same problem before

Did that exactly before incurring this problem

MervDale, 2021-08-02:
I am wondering if some of these radios have hardware problems, I have bee accessing my SD card via USB under Win 10 for quite some time and coping the most if not all firware updates to the SD  card before writing to my Eachine TX16S with no issues, as has my mate with his RadioKing TX18S under WIN 9

rotorman, 2021-08-02:
Windows 9 ?

pfeerick, 2022-04-27:
@wimalopaan Is this issue still a thing?

wimalopaan, 2022-04-27:
Yes, also with actual `main`.
After selecting `storage` from popup, it simply crashes (off). Selecting `joystick` does not crash and displays th USB-Symbol in the top-bar.
Selecting `storage` after having selected `joystick` works (sometimes at least). 

Looks like this is an initialization problem, maybe race-condition.

MervDale, 2022-04-27:
I use my USB connection to my radios time and time again without any issues for both uploading firmware, both release and nightlies as well as accessing companion,  Have you tried changing USB data cables as I said previously they are not all created equal, failing that it may well be an issue with the radio itself, I have two Eachine TX16s's and both are fine with USB.

wimalopaan, 2022-04-27:
Yes, different cables, same effect.
But using it in Bootloader all works well.

pfeerick, 2022-04-27:
I *have* experienced dropouts / non-responsive sd card when copying large numbers of small files... (e.g. yappu map data)... but not outright crashes on connect, so yeah, that is strange... Have you noticed any other patterns. like does it happen on blank models, so no lua scripts running in the bg?

wimalopaan, 2022-04-27:
I did some extensive testing and rebuilding but I can't reproduce the crash anymore!!!
So, I'll close it now.
Sorry for the noise.

closed 2022-04-27 by wimalopaan

pfeerick, 2022-04-28:
That's ok... it obviously happened at some point, but has hopefully gone away as we fixed something connected to this... otherwise you know where the bug reporter is! :grin: