EdgeTX/edgetx #537: Mixer script can't access Global Variable

wimalopaan, 2021-08-01:
The following mixer script should forward a global variable, but does not. In OpenTx it works.

```
local input = {
   {"GVarNr", Value, 0, 9, 6},
};

local output = { "send gvar" };

local c = 0;

local function run(number)
   print("c: ", c);
   c = c + 1;
   local v = model.getGlobalVariable(number, 0);
   return v;
end

return {output=output, input=input, run=run}

```

wimalopaan, 2021-08-02:
It looks like two problems in one here: the input table does not work as expected (but maybe I am wrong here) and the script can't access the gv correctly.

raphaelcoeffic, 2021-08-02:
@wimalopaan I think we have 3 issues here!

The first one I could find is fixed here: bb60d622677ecfac. Values coming from LUA mixer script (so called "LUA inputs") were not displayed properly.

Then, your script was a bit wrong, here is a version that works:
```lua
local input = {
   {"GVarNr", VALUE, 0, 9, 6},
};

local output = { "send gvar" };

local c = 0;

local function run(GVarNr)
   print("c: ", c);
   c = c + 1;
   print(GVarNr);
   local v = model.getGlobalVariable(GVarNr, 0);
   print(v);
   return v;
end

return {output=output, input=input, run=run}
```

And last but not least, it seem we have a bug in the new LUA task handling: I have the feeling that the script's `run` function is run only once at start, and not afterwards (whereby the status is ok, no errors). @jfrickmann could you please give it a try?

jfrickmann, 2021-08-02:
I will look into it!

wimalopaan, 2021-08-02:
> Then, your script was a bit wrong, here is a version that works:

You mean the name of the function parameter?

The strange thing is that it works for OpenTx.

raphaelcoeffic, 2021-08-02:
> > Then, your script was a bit wrong, here is a version that works:
> 
> You mean the name of the function parameter?
> 
> The strange thing is that it works for OpenTx.

I think the only issue was actually `VALUE` vs `Value`.

wimalopaan, 2021-08-02:
Is there a chance to consider https://github.com/EdgeTX/edgetx/issues/132#issuecomment-878767043

wimalopaan, 2021-08-02:
> I think the only issue was actually `VALUE` vs `Value`.

Ah yes, I corrected that already, but forgot to mention that here :-( 

This error results in not being able to spin the input value. There should be surely a better error message. If something is wrong, fial loud and clear.

jfrickmann, 2021-08-03:
I just tested the script above in the X9D+ simu built from main, and it seems to do what it is supposed to:
```
c:      11236
1
2
c:      11237
1
2
c:      11238
1
2
c:      11239
1
2
c:      11240
1
2
c:      11241
1
2
```
etc.

raphaelcoeffic, 2021-08-03:
> I just tested the script above in the X9D+ simu built from main, and it seems to do what it is supposed to:
> 
> ```
> c:      11236
> 1
> 2
> c:      11237
> 1
> 2
> c:      11238
> 1
> 2
> c:      11239
> 1
> 2
> c:      11240
> 1
> 2
> c:      11241
> 1
> 2
> ```
> 
> etc.

The issue was on TX16S target. I must admit I did not try that one.

wimalopaan, 2021-08-03:
On TX16s the script only called once (c: 0).

```
0,30 f_open(/home/lmeier/OpenTx/sd25.simulator/SOUNDS/en//name.wav) = INVALID_NAME (FIL 0x7f62a19f2c00)
c:      0
[W] unknown():0 - libpng warning: iCCP: known incorrect sRGB profile
11,59 ### switched to view #1
```

jfrickmann, 2021-08-03:
OK, I will try again on TX16S and get back to you!

raphaelcoeffic, 2021-08-03:
> OK, I will try again on TX16S and get back to you!

Thank you so much Sir!

jfrickmann, 2021-08-03:
When I build simulator25 for TX16S and run it, I get a seg fault when I try to open a mixer script. This happens before I can even select a script to load, so I assume that it is a problem in the GUI. I have tried to add traces and set -DDEBUG=ON, but it does not give me any info on where it blows, and I have so far not been able to nail it down with traces. I need to clear that hurdle before I can get on to looking at the Lua part. Any suggestions for how to proceed are welcome!

raphaelcoeffic, 2021-08-03:
> When I build simulator25 for TX16S and run it, I get a seg fault when I try to open a mixer script. This happens before I can even select a script to load, so I assume that it is a problem in the [...]

You should definitely run it under debugger, it will show you where it crashes.

jfrickmann, 2021-08-03:
Gdb or another debugger?

————————
Jesper Frickmann

wimalopaan, 2021-08-03:
Strange, with the latest main I do not encounter any segfaults (plattform = arch linux)

jfrickmann, 2021-08-04:
I think that I got it fixed - please see PR #552.

jfrickmann, 2021-08-04:
Regarding the simulator crash, it was deep down in the file handling. It looks like it got a NULL pointer for a string. So I tried to select the script in the SD card browser and run it there. Of course the mixer script cannot run as a standalone, but it does generate the .luac file. So I went back to custom functions, and with the .luac file present, the problem disappeared. Just sayin' ;-)
[gdb.txt](https://github.com/EdgeTX/edgetx/files/6927876/gdb.txt)

raphaelcoeffic, 2021-08-04:
> Regarding the simulator crash, it was deep down in the file handling. It looks like it got a NULL pointer for a string. So I tried to select the script in the SD card browser and run it there. Of [...]

I’ll take care of that one.

wimalopaan, 2021-08-04:
Looks like fixed now

closed 2021-08-04 by wimalopaan