EdgeTX/edgetx #538: R9 Receiver Flashing OTA FRSK files - FRSK not recognized as a valid firmware.

MartyFufkin, 2021-08-01:
I tried to flash an R9 Mini OTA receiver with an OTA firmware which ends in file extension ".FRSK".  This file extension is not recognized by the Edge firmware on my TX16S as a flashable external file and does not work.  I flashed FRSK file type successfully with OpenTx on another Taranis radio I have.

Firmware available for testing at: https://www.frsky-rc.com/r9-mini-ota/

raphaelcoeffic, 2021-08-01:
> I tried to flash an R9 Mini OTA receiver with an OTA firmware which ends in file extension ".FRSK". This file extension is not recognized by the Edge firmware on my TX16S as a flashable external [...]

Small issue here: I don‘t own any OTA receivers. So it might become quite fun to fix this properly.

MartyFufkin, 2021-08-01:
I used to do some development 20 years ago...   I may look at it and get my feet wet again.  I certainly have a work around (Taranis).  If you need a guinea pig to test for you, happy to do that for you as well.

raphaelcoeffic, 2021-08-01:
> I used to do some development 20 years ago... I may look at it and get my feet wet again. I certainly have a work around (Taranis). If you need a guinea pig to test for you, happy to do that for you as well.

If you manage to fix it, that would be of course the best/fastest option! Another would be to use a logic analyses to trace the communication on S.Port while OTA flashing with your taranis. This way, we could even find a way to simulate that module.

MartyFufkin, 2021-08-01:
Yes, it is R9 Access.   I think it is simpler than that (although I may be wrong)...   Flashing an FRK files works...  FRSK files are just not identified by the Edge firmware as being a valid firmware file so the only options are Copy and Delete.  Flash External and Flash Internal do not come up as options.

raphaelcoeffic, 2021-08-01:
> Yes, it is R9 Access. I think it is simpler than that (although I may be wrong)... Flashing an FRK files works... FRSK files are just not identified by the Edge firmware as being a valid firmware [...]

I see, this is indeed easy to fix :-)

brainbubblersbest, 2021-08-01:
> > Small issue here: I don‘t own any OTA receivers. So it might become quite fun to fix this properly.

You dont have a Single r-xsr with accst anywhere? 
It Supports OTA after updating to access. 
Time to play with the isrm in your RM. 😉

I remember of a commit on otx that whas in need to Support OTA with frsk files.

pfeerick, 2021-08-02:
A better comparison would have been OTX and ETX on the TX16S, as the TX16S may not be able to do everything update related comparied to the Taranis or X7, etc. 

On OTX with the TX16S, both .frk and  .frsk extensions add the `Flash external module` option to the popup menu. 

On ETX, with the TX16S, .frk adds the 'Flash internal module' (hm... really?!) and 'Flash external module', and .frsk doesn't add anything. 

I suspect that the internal flash entry is because https://github.com/EdgeTX/edgetx/blob/3eeca550dad17b174848f9bea59a0e5b8d45b9d5/radio/src/gui/colorlcd/radio_sdmanager.cpp#L329-L350 doesn't have include guards like https://github.com/opentx/opentx/blob/d67d1aa0c09d2f485c3ef7ca8c4fb1e9214a2ee7/radio/src/gui/480x272/radio_sdmanager.cpp#L375-L382

And the FRSKY_FIRMWARE_EXT handling looks to be entirely missing... i.e. similar to this... https://github.com/opentx/opentx/blob/d67d1aa0c09d2f485c3ef7ca8c4fb1e9214a2ee7/radio/src/gui/480x272/radio_sdmanager.cpp#L383-L399

pfeerick, 2021-08-02:
If you're brave, and willing to give it a try, #543 *appears* to add the entry, and may or may not actually flash the receiver firmware. I have nothing to test it with, so it has only gotten the "`simu` shows an entry for frsk file and says it wants to run the update process test." simulation test.

MartyFufkin, 2021-08-03:
I will give it a shot...  (although I just ordered a full set of everything Crossfire)...  nothing to lose bricking anything.

raphaelcoeffic, 2021-08-03:
> I will give it a shot... (although I just ordered a full set of everything Crossfire)... nothing to lose bricking anything.

Worst case, send me the bricked hardware, I will make good use of it ;-)