EdgeTX/edgetx #541: Cant flash any frsky receivers

Mantamatt1, 2021-08-01:
Tx16s
Latest etx main firmware.
Cant flash any frsky receivers.
I have set external bay to ppm (and everything else) and set up to flash like I always do but...
It trys to flash then a message pops up saying "firmware update error data refused"
Receiver goes from flashing red indicating power.
Then off. Then flashing green and that's when the message pops up.
Receivers I have tried are s6r. Rxsr and r9mm.
I put the receiver flash files in a qx7 running otx and flashed straight away.

raphaelcoeffic, 2021-08-02:
> I have set external bay to ppm (and everything else)

This is not necessary at all! You can just leave it OFF. Flashing from the module bay should turn the power on when required (it needs to reset the receiver to trigger the bootloader).

Mantamatt1, 2021-08-02:
I had also set to off and still have the same message

commit 6a202015 (2021-08-02): Fix flashing firmware over S.Port (modules / receiver).

Resolves #541
closed 2021-08-02 by raphaelcoeffic