EdgeTX/edgetx #546: Add USB telemetry mirror (Opentx PR #8650)

wimalopaan, 2021-08-02:
Add the PR https://github.com/opentx/opentx/pull/8650

raphaelcoeffic, 2021-08-02:
Not bad! But I think we can make this better by leaving this as-is and adding settings to map serial ports to functions. So as you can choose now to have telemetry mirror on a normal serial port, you could assign the function “telemetry mirror” also to the usb serial. We will need this capability for mavlink as well.

raphaelcoeffic, 2021-08-02:
> Not bad! But I think we can make this better by leaving this as-is and adding settings to map serial ports to functions. So as you can choose now to have telemetry mirror on a normal serial port, [...]

These assignment should be per model, of course, as you might want different things depending on the model.

wimalopaan, 2021-08-02:
You mean, we have in all radios the function "usb serial" to which one can assign "SBus in", "Telem mirror", "LUA", etc?

Wow, that would be real cool. One has to compile the firmware with USB_SERIAL=YES?

Looks like I missed the USB_SERIAL option complety ... what can I do with that in the current state of the project? Only debug?

raphaelcoeffic, 2021-08-02:
> You mean, we have in all radios the function "usb serial" to which one can assign "SBus in", "Telem mirror", "LUA", etc?
>
> Wow, that would be real cool. One has to compile the firmware with [...]

That is a plan, it does not work yet ;-) But you can today have debug via USB serial by enabling it, yes. It's just, you cannot choose what it is used for, whereby you can for the normal serial ports.

wimalopaan, 2021-08-02:
> That is a plan, it does not work yet ;-) But you can today have debug via USB serial by enabling it, yes. It's just, you cannot choose what it is used for, whereby you can for the normal serial ports.

Simply awesome!!!

Meldinov, 2022-04-25:
CMON, OpenTX now offiaclly adopted this! :P

rotorman, 2022-04-25:
@Meldinov Have you tried latest EdgeTX already?

[image]

Meldinov, 2022-04-25:
> @Meldinov Have you tried with latest EdgeTX?
> 
> [image]

then why is this still an open issue?

closed 2022-04-26 by raphaelcoeffic