EdgeTX/edgetx #547: Forward Programing for Spektrum Rxs

Mikey1968, 2021-08-02:
So I know Opentx has made a forward programing lua but seems like they never finished it, would you or someone make a working forward programing for spektrum rx users

raphaelcoeffic, 2021-08-02:
> So I know Opentx has made a forward programing lua but seems like they never finished it, would you or someone make a working forward programing for spektrum rx users

Wasn't it done by the MPM guys? Frankly, I don't even know. But yes, why not, would if nice if someone would update it.

pfeerick, 2021-08-03:
Forward programming is done via the Lua Script provided as part of the MPM - https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts

Specifically, download the `DSM FwdPrg.lua` to the `/SCRIPTS/TOOLS` directory of your transmitter's SD card, and it will be visible in the SYS -> Tool screen. As well as any of the other scripts you add.

Mikey1968, 2021-08-03:
Where do you download it at

pfeerick, 2021-08-04:
Um... At the link I gave? It's the first file in the list... click on the file listing, then you can right click on the 'Raw' button over on the top right of the file view, and download the script ready to put on the SD card.

Mikey1968, 2021-08-04:
I don't see no raw button. I clicked on the first file but see no where to
download it

Mikey1968, 2021-08-04:
I see view RAW but there's no download

pfeerick, 2021-08-04:
Right click, and choose save link as or equivalent for your web browser.

pfeerick, 2021-08-04:
[image]

pfeerick, 2021-08-11:
@Mikey1968 How did you go with this... did you get it working?

Mikey1968, 2021-08-11:
I haven't tried it yet. Need to buy a Rx to try it .