EdgeTX/edgetx #548: Add a "Close Log" special function to make the next flight begin a new log file.

Miami-Mike, 2021-08-02:
Add a special function called "Close Log" to close and rename the current log file so that a new log file will start with the next flight. Please see [this discussion on the OpenTX GitHub forum](https://github.com/opentx/opentx/issues/7752).

lshems, 2021-08-02:
How would we determine the new logfile name? Sequence number in file name? Then also rename the first log with sequence number. And make sure you dont run into trouble with models named model01 and model02, so sequence number should be after date addition?

raphaelcoeffic, 2021-08-02:
@Miami-Mike I might have missed the point, but wouldn't it be perfectly enough be able to stop/re-start the log, thus creating a new file? (numbered prefix? new date? whatever)

MikeBland, 2021-08-02:
In erskyTx I have an option when logging of keeping the same file for the day, or starting a new file each time logging is started. In the second case, I simply include the time it is started as part of the logfile name.

raphaelcoeffic, 2021-08-02:
> In erskyTx I have an option when logging of keeping the same file for the day, or starting a new file each time logging is started. In the second case, I simply include the time it is started as part of the logfile name.

As usual, sounds like a sound solution. Not sure the setting is absolutely necessary, it could be this way by default, but then you don’t see all of them on the same screen with companion.

lshems, 2021-08-02:
Although it would complicate Lia log file viewing on the radio, why not. I detest options

raphaelcoeffic, 2021-08-02:
> Although it would complicate Lia log file viewing on the radio, why not. I detest options

Betaflight uses a new file after each "arming" of the board, which basically boils down to the same. Not sure anyone complained about that ever.

lshems, 2021-08-03:
What is the trigger for splitting? Why not trigger it on the first write action. You need to activate the special function 'log' with a switch, so that switch will trigger the split?

lshems, 2021-08-03:
There is an unwanted effect of using a time label instead of sequence label. No way to know the filename now to be able to load it in lua.

Please consider using a sequence number instead.

pfeerick, 2021-08-03:
Trigger for splitting is logging turned on again. i.e. the filename is determined each time logging starts.

lshems, 2021-08-03:
Percect

raphaelcoeffic, 2021-08-03:
> There is an unwanted effect of using a time label instead of sequence label. No way to know the filename now to be able to load it in lua.
> 
> Please consider using a sequence number instead.

You should "scan" for files in the directory, and parse the filenames. Did you previously just try to open a file named according to the pattern, using the same method (without checking there is actually a file to open)?

raphaelcoeffic, 2021-08-03:
> Trigger for splitting is logging turned on again. i.e. the filename is determined each time logging starts.

Exactly, and when the file is empty while opening, it will dump the headers and go on. Otherwise, it will continue the existing file with the same headers (did not check how it knows which fields to use...).

lshems, 2021-08-03:
There was no scan so indeed it was searching on dates and see if there is a file.

What to use to scan?

raphaelcoeffic, 2021-08-03:
> There was no scan so indeed it was searching on dates and see if there is a file.
> 
> What to use to scan?

Wait a sec: which function did you sue to read the files? I guess those coming from LUA directly, no? Or is that something provided by our API?

lshems, 2021-08-03:
There is no file and directory search available, as they implemented a crippled Io library.

So you need to know the name of a file and its location, otherwise you can't load it in LUA edge.

Numbered files are thus no issue, as the original telem001.lua, telem002.lua, etc.

lshems, 2021-08-03:
So no direct MUA Io library available, only a few read and write functions.

lshems, 2021-08-03:
I even made a copy file function in lua using read, create and write. Would be nice if available native

raphaelcoeffic, 2021-08-03:
> I even made a copy file function in lua using read, create and write. Would be nice if available native

Ok, I’ll check what is already there.

jfrickmann, 2021-08-03:
I agree that if we start renaming the log file, then we would also need support for listing directory content. In SoarOTX, I construct the log file name from the model name and the current date, and that would stop working if we start renaming it.

raphaelcoeffic, 2021-08-03:
> I agree that if we start renaming the log file, then we would also need support for listing directory content. In SoarOTX, I construct the log file name from the model name and the current date, and that would stop working if we start renaming it.

Do you push to the same file as the normal logs? Otherwise we can safely continue to use your own file names.

However, I do agree we need to have file operation that allow for iterating/listing the content of a directory. If this is not already there, it needs to be added. I believe LUA should have these routines, but I suppose we will have to provide some API for listing a directory (step-by-step) if this isn't included yet.

raphaelcoeffic, 2021-08-03:
> > I agree that if we start renaming the log file, then we would also need support for listing directory content. In SoarOTX, I construct the log file name from the model name and the current date, and that would stop working if we start renaming it.
> 
> Do you push to the same file as the normal logs? Otherwise you can safely continue to use your own file names.
> 
> However, I do agree we need to have file operation that allow for iterating/listing the content of a directory. If this is not already there, it needs to be added. I believe LUA should have these routines, but I suppose we will have to provide some API for listing a directory (step-by-step) if this isn't included yet.

lshems, 2021-08-03:
I think there is a miscommunication.

Jesper and I and others have written lua scripts to view, graph,.... The logfiles.

We can do that because they are named with the model name and date, and you simply open all possible dates in a range looking for files.

We don't create not push anything ourselves.

So if you add a time stamp, we would have to scan for 24*59 possible time labels within a day to find the files.

If using a 3 digit sequence number, we would just load the first, second, ... Until we find none.

If we have a directory listing function, we wouldbt care how you name them,we'll find them.

raphaelcoeffic, 2021-08-04:
> I think there is a miscommunication.
>
> Jesper and I and others have written lua scripts to view, graph,.... The logfiles.
>
> We can do that because they are named with the model name and date, and you [...]

Would something like that be what you're looking for? https://www.lua.org/pil/29.1.html
It seems LUA does not offer it by default, so that we have to code our own (following that example?).

lshems, 2021-08-04:
We need to invite max paperno. He had this all running, bit for some reason it didn't make it into opentx

raphaelcoeffic, 2021-08-04:
> We need to invite max paperno. He had this all running, bit for some reason it didn't make it into opentx

I have no clue how to contact him, but if you do, feel free to do so!

mpaperno, 2021-08-04:
Hi guys, you rang?   :-)

If I understand correctly that you're looking for file system operations from within Lua scrips, then yes indeed I had a PR for this but like many things it never made it into OTX.

https://github.com/opentx/opentx/pull/6148

With examples and docs and all.

```
@function fileSystem(operation, path [, to_path | get_file_info])
Performs various file system operations.
@param operation (string) (required) The action to perform. Valid operations are:
     * `stat` - Checks if file/directory exists and optionally returns additional extended information about it. Returns (0) if exists.
     * `mkd`  - Create a directory.
     * `ls`   - Return a directory listing.
     * `ren`  - Rename or move a file or directory.
     * `cpy`  - Copy a file.
     * `del`  - Delete a file or directory. (Directory has to be empty.)
 ```

It uses the existing C `fatfs` file functions to implement the actual features.  Assuming ETX still uses a file system for SD access (seems like a fair assumption :) then it should just work, or require very little modification.  In fact IIRC the Lua [`io` module in OTX](https://github.com/opentx/opentx/blob/4c296fa9b8f18e3c7af713f0d48c815de50ac98b/radio/src/thirdparty/Lua/src/liolib.c) was heavily modified to use `fatfs` functions directly (also why it is limited in what is exported).

Note the supporting fix for the simulated `fatfs` features, as I recall that was required for the new function to work in the simulator.

AFAIK Lua doesn't have "native" file system ops like directory listing. There's [LFS](https://keplerproject.github.io/luafilesystem/) but again this would all need to be modified to use `fatfs` (I assume), and besides it seems like overkill.

LMK if you guys have questions or want help integrating this.  Oh, and cool project... I'll have to poke around! :)  Not much flying lately for me.  I have a ton of other stuff that never made it into OTX (many I never even submitted because I got too far ahead of the releases or other refactors).

Cheers,
-Max

lshems, 2021-08-04:
Super, max. I'm sure you love the approach taken on edgeTX

raphaelcoeffic, 2021-08-04:
@mpaperno just had a look at the PR, which looks quite good to me! Only aspect I would change would be to use an iterative method for “ls” operation, thus avoiding to return a table which can have a potentially very big size (mem constraints, again…). This way, scripts could have the same memory consumption regardless of the number of files in directory.

raphaelcoeffic, 2021-08-04:
@lshems @jfrickmann would the proposed code suit your purposes? This looks like something essential to me. Don’t know why this has never been included.

raphaelcoeffic, 2021-08-04:
> Don't ask questions on why things never made it. Just look ahead instead.

Right, let me rephrase: I don’t see any reasons for not including this.

raphaelcoeffic, 2021-08-04:
> LMK if you guys have questions or want help integrating this. Oh, and cool project... I'll have to poke around! :) Not much flying lately for me. I have a ton of other stuff that never made it into OTX (many I never even submitted because I got too far ahead of the releases or other refactors).

Thx a lot @mpaperno! I cherry-picked the code in into a branch based on current `main`. There were a few conflicts, but nothing dramatic so that I think it can be made to work pretty fast.

jfrickmann, 2021-08-04:
Awesome!! Thanks @mpaperno  and @raphaelcoeffic 🥇

commit 5b5ea807 (2021-08-03): Use the current time to generate the log filename

Resolves #548
PR #550 (2021-08-31): Use the current time to generate the log filename
This changes the log file name format from `modelnamexxxxxx_YYYY-MM-DD.log` to `modelnamexxxxxx_YYYY-MM-DD-HHMMSS.log`. This effects that basically every new flight will create a new log file (with its own header, etc...).

Resolves #548
closed 2021-08-31 by raphaelcoeffic