EdgeTX/edgetx #557: Model Checklist icon not available

stefaanbolle, 2021-08-04:
I have compiled the latest commit a2b8b4e and installed it on my TX16S.
All my models have a checklist. The checklist is correctly displayed when selecting a new model.
However I do not have a model notes icon in the touch menu. It is not available for any model with a checklist.

pfeerick, 2021-08-05:
Sorry, I can't reproduce that. I just grabbed the binaries for that build (https://github.com/EdgeTX/edgetx/suites/3410410412/artifacts/80622771) and have both the checklist appearing on startup/model select and the model notes icon showing (second icon) showing on my TX16s. 

I would ask what the file was called, but if it's showing as the checklist, from what I can remember off the code of the top of my head it, the model notes file is detected, so the menu icon should also be there, as it uses the same test.

stefaanbolle, 2021-08-05:
The checklist is in the MODELS folder and has rhe same name as the model name. I compile my own binaries since I use internalgps (modded my TX16S with a BN220).

stefaanbolle, 2021-08-05:
Uploaded the file @pfeerick provided in the post above and than I have the extra button. When I compile from main repo I do not have it.
Any reason why? Would also like to use the internalgps option

pfeerick, 2021-08-05:
Damn... 

what build flags are you using... as internalgps isn't working for me now... which is annoying now as I have built it before as I tested the GPS out with my TX16s also... just haven't gotten around to fitting it internally. 

I thought it was:

`cmake -DPCB=X10 -DPCBREV=TX16S -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release ../`

but it's not listed as enabled in the built simu or firmware...

stefaanbolle, 2021-08-05:
These are the options I use
`-DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DINTERNAL_GPS=YES -DCMAKE_BUILD_TYPE=Release`

pfeerick, 2021-08-05:
Ok, something must be broken - perhaps the cmake 3.13 update broke something :-/

This is what I get with the same flags:
[image]

but on your binary, I also have the `internalgps` flag as well.

btw, that particular binary will *not* have the model icon, as that particular commit (09a65807) is from from 14 days ago, and adding of the model icon to the carousel was only done five days ago in (645afc11767452049e7d5e336fe99f3e64501d23), and I updated the icon a few hours later in (4ec62d71b61a26d7a51a3233a53c86c829a50df3).

stefaanbolle, 2021-08-05:
I git pull the main repo. What do I need to do to include your 645afc1 commit

pfeerick, 2021-08-05:
You shouldn't need to do anything if you have pulled the repo. Try `git fetch` to make sure it knows the current state of the github repo, and then `git status` to see what branch you are on, and the status of your local copy... it should come back indicating you are on `main` and that it is up to date...

```
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

pfeerick, 2021-08-05:
Unless you have better luck than me, it seems like since the cmake version requirement was bumped, internalgps fails to be included when requested, so I ended up doing `git reset --hard ab1dd80` to jump back to the last commit that works for me.

raphaelcoeffic, 2021-08-05:
> Unless you have better luck than me, it seems like since the cmake version requirement was bumped, internalgps fails to be included when requested, so I ended up doing `git reset --hard ab1dd80` to jump back to the last commit that works for me.

Let's solve this ASAP via Discord ;-)

stefaanbolle, 2021-08-05:
was on 2.4 branch. Now on main branch.
Compiled without errors.
Internal GPS seems to be active (at least the icon on top is there)

stefaanbolle, 2021-08-05:
Was too quick, in the statistics/info screen, the option internalgps is not there. Probably lost during compilation although enabled in the params

raphaelcoeffic, 2021-08-05:
> Was too quick, in the statistics/info screen, the option internalgps is not there. Probably lost during compilation although enabled in the compile parameters
> So same issue as you have i main branch with latest commit

Issue highjacking! 😅

Your problem has nothing to do with the Checklist any more ;-) We're solving it currently. It is related to some recent cmake changes...

stefaanbolle, 2021-08-05:
Ok, thanks! Do you want me to close this issue?

pfeerick, 2021-08-05:
Yeah, might as well... Throw up a new one if you can't build internalgps in the next day or so... 
If you really want it right now... there are to options... 

1) roll back to this commit... (three commits ago, at time of writing), and you should be good to rebuild and internalgps will be there. 
```
git reset --hard ab1dd80
```

2) You could edit `radio/src/targets/horus/CMakeLists.txt` after having run cmake (but before running make), and removing 
`set(INTERNAL_GPS OFF)` at around line 65... and then running make to build your firmware. 

To be honest, option one is the simplest, and once the issue is fixed, you can then just pull the current version of the repo without any problem.

raphaelcoeffic, 2021-08-05:
@stefaanbolle I pushed this, which should fix the pb: 49bec9110d974431

closed 2021-08-05 by raphaelcoeffic

pfeerick, 2021-08-05:
Yup, fixed it for me... ;)

stefaanbolle, 2021-08-05:
For me too, thanks guys!