EdgeTX/edgetx #558: Enhancement: Rethink how SD card contents are distributed.

Miami-Mike, 2021-08-04:
Presently a large variety of SD card content zip files are offered for many different radios, but if you examine their contents you'll find that there really isn't much difference between them. If you consolidate them then they make up a single set of files that will work with any radio. There will be some folders and files that won't be useful for some radios, such as a WIDGETS folder for a Taranis radio, but the presence of that particular folder does no harm, and with a size of 41.2 KB it consumes an insignificant amount of room on an SD card.

So I suggest distributing a single SD card contents zip file with a complete set of files that work with any radio. It would benefit anyone who owns or works with different radio types because it would both save the need for multiple downloads, and save the need to change the SD Structure path setting in Companion to work with a different radio type.

lshems, 2021-08-04:
Manage sdcard content better #187

MervDale, 2021-08-05:
I'm wondering if this is an issue these days with the size and price of SD cards these days eg, I have 1450 files on my Opentex 2.3.14 SD card and it takes up less than 40 MB....  AUD $9.00 for a SanDisk Ultra 16 GB

pfeerick, 2021-08-05:
Size isn't really the issue... it's more how long it takes to sync/update things... i.e. it takes a lot longer to do a SD sync with all seven or so voice packs... and it's a lot faster if I only have `EN` on ;) Particularly because these are lots of small files, which completely buggers up write speeds.

raphaelcoeffic, 2021-08-05:
> Size isn't really the issue... it's more how long it takes to sync/update things... i.e. it takes a lot longer to do a SD sync with all seven or so voice packs... and it's a lot faster if I only [...]

I thought we already agreed on the sound packs being a different thing, meaning that you select the language and the Flasher would “stitch” your image together. For those who don’t want to use the flasher for this, they would download the “generic image”, plus their special sound pack (by language? Nothing specific to radio here).

pfeerick, 2021-08-05:
Yup... but the issue being raised here is I think more along the lines of why are there four base images, rather than just one image that you can download. It's *not* one of size. ;) Especially as, for instance, a B&W radio will basically ignore all the extra colour files, because it doesn't know what to do with them. Which then also partly feeds back into lshems thoughts as to making it so that you could basically have version folders on the SD card, making it easy to switch between pack 'versions' ... Whether that is needed, what it would look like or if it is practical is up for discussion ;)

raphaelcoeffic, 2021-08-05:
Ok, then let’s agree that we can provide a single image, in theory. The added files are not worth discussing, as we have seen, size on SD is not an issue. Confused users is.

pfeerick, 2021-08-05:
In that case, it would probably just be a case of mashing the four images together, and providing that as a "don't think, just unzip", for those who don't want to use Flasher (which for this I think totally should - how hard is it to pick the name of your transmitter, then tick the language for the voice pack, and then pick the SD card drive), or work out which image they need.

lshems, 2021-08-05:
The main issue is for users with multiple radios to manage different Sdcard versions AND customisations.

Two issues.

The first is simply merging all radio sdcrad content for all radio types on one Sdcard image. Let the radio determine where it needs to get his specific files.

The second is the user added content. Scripts, images, voicefiles, etc 

You don't want that altered when updating the firmware.

lshems, 2021-08-05:
Using flasher is fine, no issues as such. Just allow flasher to ADD a radio type specific data set to the SDcard, while it won't touch the already loaded radio type.

I switch sdcards regularly just to get my lua scripts quickly on another radio.

Other people do the same for voice files and I don't know what.

pfeerick, 2021-08-05:
Maybe a feature request is needed over at https://github.com/EdgeTX/flasher then, for ability to copy a folder to the SD card? 😉 

Although, at the end of the day, if that is *all* you are doing... it is probably quicker/easier to open your file browser, select and copy your folder... than it is to open Flasher and find the options there. I personally think it would only make sense if you were provisioning a new card, or updating the card generally, and additionally wanted a folder of stuff already on your computer to be added on as well.

lshems, 2021-08-05:
I think we talk next to each other.

If I can take an Sdcard, and use flasher to get it ready for an x9lite, I would like to be able to take the same card and 'flasher' it to an X10. Both should now work with the SDcard.

My personal sound files, images, scripts, I want to use them on all radios I use the card with.

So, therefore my suggestion to make a folder per radiotype, and flash the radio Sdcard content to that folder instead of root. Let the radio use that folder as curdir and nothing needs to be changed for now.

Then, in a later stage, we van think about how to organise user specific files like sounds and scripts. 

For scripts, it is not even a problem to cater for this with a proper LUA loader script that will check for itself where to look for the scripts to run based on user preferences.

Miami-Mike, 2021-08-05:
> pfeerick:
> 
> Size isn't really the issue... it's more how long it takes to sync/update things... i.e. it takes a lot longer to do a SD sync with all seven or so voice packs... and it's a lot faster if I only have `EN` on ;) Particularly because these are lots of small files, which completely buggers up write speeds.

I didn't mention it in the OP but I was thinking that if you want to offer a choice between multiple downloads then to me it would make more sense to offer a choice of languages rather than a choice of radios. A user can own or work with multiple radio types but will rarely have more than one favorite language.

You could offer one universal SD card contents file and a separate download of your choice of SOUNDS folders.

raphaelcoeffic, 2021-08-05:
> You could offer one universal SD card contents file and a separate download of your choice of SOUNDS folders.

Ok, so it seems this is agreed on.

gagarinlg, 2023-03-18:
the only thing preventing us doing this is currently that the LUA scripts for B&W and Color LCD are not identical, because not all scripts are compatbile to both worlds. Any ideas how to solve this?
Maybe subfolders for B&W and Color LCD where the specific scripts are placed and they are then not runnable from the other kind of radios?

lshems, 2023-03-18:
No. Please not. Scripts are compatible and attention should be given to increase compatibility.

I declare variables locally and assign the existing global 'constant' or what is missing, for all constants to work on all radios. So for instance the GREY constant doesn't exist on Color but does on BW. In my scripts I then declare:

Local GREY = GREY or LCD.color(backggroundcolor) or something like that. (On my phone with no pc at hand to check the exact code I use).

This GREY constant can then correctly be used in all LCD functions as a constant, on any radio.

All scripts are essentially the same, and perhaps we should work on unifying the API for scripts.

I ONLY write onetime scripts, to create a run, background and init function, with added constants and functions to cover for missing functions and constants in BW resp.color radio's, to get an identical function set to be used for all scripts.

I then load those to telemetry scripts and widgets, without any further care, for both BW and Color radio's.

Actually, the different script types are unnecessary. All scripts can run with a universal run,init and background function,on both BW and Color radio's, without alterations.

Perhaps first work on that? 

Then there needs to be only one place to store scripts defining the run, background and init scripts, and the GUI will allow using those scripts as widgets, onetime, telemetry, or whatever...

lshems, 2023-03-18:
It would not even be problematic to do what you suggest by the way. Smart LUA scripters will simple write a loader script and load their 'universal' script from all those places. And then tackle the compatibility improvements in another issue.

gagarinlg, 2023-03-18:
I do not want to sort all scripts into radio type sepcific sub folders, only those that are not able to run on both types. All scripts that can adapt will be placed where they are now.
Maybe it would be good making a chapter for the EdgeTX Lua documentation on how to make scripts cross platform compatible?
@phileville ?
@JimB40 ?

gagarinlg, 2023-03-18:
We will not have universal compatibility. The color radios are much more capable and access to the clolr LCD UI elements will come. So when those new features are to be used, it will not work on B&W

lshems, 2023-03-18:
Yeah, well, I see that the trend is to go fully Color and touch. A choice.

For who's interested, some ideas to get scripts compatible for BW and colour. 

https://www.rcgroups.com/forums/showthread.php?2983474-Creating-a-graphing-script-for-OpenTX-using-LUA

gagarinlg, 2023-03-18:
> Yeah, well, I see that the trend is to go fully Color and touch. A choice.

B&W are and will be supported and there will be new radio releases with B&W screens and maybe even more RAM.

Color LCD touch screens just open up a lot of posiibilitiesregarding UI and different screen formats. Adding API calls to make well integrated LUA UIs, that look like native UIs and saving a lot of computing power at the same time, sound like a good way forward to me.
The UI code of EdgeTX can then partially redraw only those parts, where something changed, instead of doing full screen redraws all the time.

lshems, 2023-03-18:
Don't get me wrong. I do appreciate and understand the directions you are going. The edge team has really revived a more or less dead project to its full potential. I'm really impressed by what has been achieved. Both technically as well as organisation wise. 

I am just still hungry for a lua API allowing to access all radio settings, together with a solid API for user interactions, so screen output and button inputs.

These should be as similar as possible to allow for at least a certain class of scripts to be portable between all radios. So far, apart from touch input, with some tweaks, this is already the case. As all my scripts and menus are already running on all radios.

The reason they should be similar is the same reason you want models to be exchangeable: create an eco system for models and scripts.

This won't happen if the effort to make models or scripts work on every radio type is to cumbersome...