EdgeTX/edgetx #559: ModelId Select not sent to module when updated in model config (CRSF)

pkendall64, 2021-08-04:
Using EdgeTX V2.4.0 release #09a65807 on Radiomaster TX16S Max

Whilst working on a change for ExpressLRS where we can use the model id as an index for different module configuration parameters as well as model match on the RX, I ran into an issue where the model Id is not updated on the module until I select a different model and then back to the model that I updated the model Id in.

I would think that the modelId select sub-command would also be sent if the model id was updated in the model configuration page.

pfeerick, 2021-08-05:
Oh, noice! That was probably the only thing I didn't like about ELRS... not having ModelMatch (in Spektrum parlance)... and I see there is some other nice stuff already planned for v1.1... good times ahead! ~~Is there anywhere I can see this or is it still a work in progress?~~ ([Found it](https://github.com/pkendall64/ExpressLRS/commits/use-modelid-to-select-TX-configuration), I stalked your github page 😆 )

pkendall64, 2021-08-05:
It's still a work in progress! 🤣

pfeerick, 2021-08-05:
Just to make sure I'm looking at the right thing... is this the user settable "receiver id" (which then seems to get called modelId, because, why the heck not?)

https://github.com/EdgeTX/edgetx/blob/a2b8b4ecba7c253448c0c70a2bd3c958512b6861/radio/src/gui/colorlcd/model_setup.cpp#L910-L914

pkendall64, 2021-08-05:
Yes, that is the setting. I'm not sure if it's just the TX16S. I think it might just be crossfire i.e. module related. CapnBry mentioned that using a multiprotocol module it changed whenever the value was updated in that settings page.

pfeerick, 2021-08-05:
Ah, yes, of course... I saw that myself with the MPM and D16 - would lose lock as soon as I changed the number (i.e. worked as expected).  Nice, that suggests it probably is specific to crossfire handling then, which is good in the respect it only needs to be fixed once ;)

OK, I think I see the root problem... the model ID is only sent in the second frame (assuming we're counting from zero here) at  `CRSF_FRAME_MODELID` ... so doesn't get updated again until the module is power cycled as part of a model select change and CRSF is re-initialised. 

https://github.com/EdgeTX/edgetx/blob/a2b8b4ecba7c253448c0c70a2bd3c958512b6861/radio/src/pulses/crossfire.cpp#L85-L88

pkendall64, 2021-08-05:
Yes, thats the behaviour that I'm seeing here.

pfeerick, 2021-08-05:
I leave it to @raphaelcoeffic  to some up with something prettier than the highlighted changes... `simu` is telling me `createCrossfireModelIDFrame()` is being called as soon as I change the model ID...

[image]

raphaelcoeffic, 2021-08-05:
> I leave it to @raphaelcoeffic to some up with something prettier than the highlighted changes... `simu` is telling me `createCrossfireModelIDFrame()` is being called as soon as I change the receiver ID...
> 
> [image]

Thx a lot for the background research! I’ll take care of that one.

pkendall64, 2021-08-05:
Thanks guys, so quick!

pfeerick, 2021-08-05:
When living on the edge... 🤣

pkendall64, 2021-08-05:
BTW, my ELRS branch has been updated and works now. Currently you just have to specify the model number when compiling the RX firmware. I'm updating the RX wifi page to it can be configured from there soon.

raphaelcoeffic, 2021-08-05:
> When living on the edge... 🤣

Aerosmith? 😅

pfeerick, 2021-08-05:
> works now. 

That sounds ominous! 😆 That sounds perfect. If you can set it from the wifi page, will it be configurable via the lua script also then?

pfeerick, 2021-08-05:
> Aerosmith? 😅 (not to myself: another band I need to see live before they die)

You've clearly forgotten the name of the project (but have good music taste, so are forgiven) 😝

pkendall64, 2021-08-05:
> > works now.
> 
> That sounds ominous! 😆 That sounds perfect. If you can set it from the wifi page, will it be configurable via the lua script also then?

No, I don't think we can do that. Currently my thoughts are default configured when flashing, updatable via wifi and probably a betaflight passthrough command. But I will look into a LUA command to tell a connected receiver to update it's modelId.

PR #560 (2021-08-05): [color] Set proper CRSF module state when model-ID has changed
Resolves #559
commit 356449ce (2021-08-05): [color] Set proper CRSF module state when model-ID has changed

Resolves #559
closed 2021-08-05 by raphaelcoeffic