EdgeTX/edgetx #563: Add color constant table to LUA API

JimB40, 2021-08-06:
This is needed for Themer but in broader view it will allow to access any C++ constant available in LUA API

So my proposal is to add it to LUA _ENV variable

Example:
_ENV.CONSTANTS
so
_ENV.CONSTANTS.OS_COLOR_PRIMARY1 or _ENV.CONSTANTS.BLACK or _ENV.CONSTANTS.SMLSIZE is accessible this way

jfrickmann, 2021-08-06:
They have already bee added in #540:
https://github.com/EdgeTX/edgetx/blob/a93ac741727e1bb1c8049f56931c25c24ecbb16f/radio/src/lua/api_general.cpp#L1908

raphaelcoeffic, 2021-08-07:
> They have already bee added in #540:
> https://github.com/EdgeTX/edgetx/blob/a93ac741727e1bb1c8049f56931c25c24ecbb16f/radio/src/lua/api_general.cpp#L1908

I think @JimB40's original issue was that he cannot access these constants with `_ENV['some constant']`. Possibly, the constants in the library table are not added to that (don't know why yet).

jfrickmann, 2021-08-16:
So I do not quite understand this. In Lua, _ENV is a table that holds the global environment, i.e. all global variables. So any reference to a global variable is an entry into _ENV, and only if you hide a global variable by declaring a local of the same name, would you ever need _ENV. The other use of _ENV would be to do trickery in the C API functions to change the global environment for loaded chunks. Please see: https://www.lua.org/manual/5.2/manual.html#2.2

raphaelcoeffic, 2021-08-16:
> So I do not quite understand this. In Lua, _ENV is a table that holds the global environment, i.e. all global variables. So any reference to a global variable is an entry into _ENV, and only if you hide a global variable by declaring a local of the same name, would you ever need to refrence _ENV explicitly. The other use of _ENV would be to do trickery in the C API functions to change the global environment for loaded chunks. Please see: https://www.lua.org/manual/5.2/manual.html#2.2

I’m really not sure this is working like this with custom libs, this probably requires testing a bit more and diving into LUA code.

jfrickmann, 2021-08-16:
But I do not understand what problem we are trying to solve here? Is it that Robert cannot access the new theme color constants? From the first post here, it appears that you are looking for OS_COLOR_SOMETHING. But from your [post](https://github.com/EdgeTX/edgetx/pull/540#issuecomment-893221092) in #540 I assumed that you wanted them to be COLOR_THEME_SOMETHING. Is that the real problem? I will add a clarification to #540 of exactly what I have done with colors.

raphaelcoeffic, 2021-08-16:
> But I do not understand what problem we are trying to solve here? Is it that Robert cannot access the new theme color constants? From the first post here, it appears that you are looking for [...]

Right, that needs to be clarified first!

jfrickmann, 2021-08-16:
> Right, that needs to be clarified first!

Please see https://github.com/EdgeTX/edgetx/pull/540#issuecomment-899458434

JimB40, 2021-08-23:
@jfrickmann No problem with accessing new constants or old ones at all :)
Problem with dynamic access (string evaluation)
Commented in #540.

jfrickmann, 2021-08-23:
OK, I get what you mean now with taking a string and using it to look up a Lua variable. I don't know of a way to do it, but I will see if there is a way. For now, we only have 11 theme colors, and I don't see it as a problem. But if it really is a problem, then maybe we can make a Lua API function to do it!

JimB40, 2021-08-23:
That's why I proposed to add C++ constants available for LUA to _ENV if possible.
So we can
1) _ENV.LCD_W or _ENV.COLOR_THEME_PRIMARY1
2) _ENV["LCD_W"] or _ENV["COLOR_THEME_PRIMARY1"]
@raphaelcoeffic do you think it's possible without touching LUA interpreter?

jfrickmann, 2021-08-23:
This will do it:
```
t = _ENV[ "COLOR_THEME_PRIMARY1" ]
```