EdgeTX/edgetx #564: AIO FC - Sensors not working after quad power cycled

AndreaCCIE, 2021-08-06:
**Describe the bug**
I have a bunch of sensors working with my whoops (AIO FC - FrSky SPI).
If I disconnect the lipo and reconnect it, some of them show in RED on the telemetry page. The standard ones are ok. Rediscover the sensors doesn't change anything.
Turning off and on the radio, fixes the problem and the sensors work again without having to rediscover them.

**To Reproduce**
Go to the temetry page, disconenct and reconnect the lipo.

**Expected behavior**
Sensors should be available after changing lipo

**Screenshots**
First Lipo connected:
[image]

Second Lipo connected:
[image]

**General information**
 - EdgeTX 2.5.0 77f6b5a5
 - TX16S

pfeerick, 2021-08-06:
I've also got a TX16S, and a Mobula7 with onboard Frsky-type receiver... running the same nightly build, it will only do that if I have an intermittent battery connect... once the battery is connected properly it settles down and shows all the telemetry.

AndreaCCIE, 2021-08-06:
Hi @pfeerick, I disconnected one lipo and connected another one.
I left the whoop to sit on the bench for 2 minutes, but the sensors didn't come back.
I noticed the same behaviour with all my 3 whoops running on AIO SPI. No issue with CRSF instead.

[image]

pfeerick, 2021-08-06:
Damn. Well hopefully someone else can reproduce it, or might have some ideas on something to try.

AndreaCCIE, 2021-08-06:
Ok thanks, let me know if there's any other test you want me to run.

gagarinlg, 2023-05-05:
are there any news on this one?

pfeerick, 2023-05-05:
I still get this occasionally with 2.8.x on my main TX16S with ELRS, and I can't place a finger on it - I usually get telemetry from the TX module, but not from the FC... which makes me think this is really a betaflight issue still... as telemetry is clearly working since you wouldn't get the tx module telemetry (i.e. like in the second picture in the initial post).