EdgeTX/edgetx #566: Model Timer 1 changes from On to Start when moving from OpenTX

sohamsp14, 2021-08-06:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

When a timer is set in OpenTX as On with a switch. After conversion to EdgeTX model, the setting changes to "Start" and timer keeps counting down until it's changed to "Start" again. Also the Persistent option gets filled with gibberish. 

**To Reproduce**
Steps to reproduce the behavior:
Create Timer 1 on a model in OpenTX. 
Set name
Set mode = ON and a switch position. 
Set time
Set countdown = haptic , 30s
Set persistent = Manual Reset

Now flash EdgeTX and let the models get converted. In the new EdgeTX model, Mode becomes Start and sometimes there is gibberish in the Persistent field. 
**Expected behavior**
A clear and concise description of what you expected to happen.
Timer settings should remain consistent after moving to EdgeTX
**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
2.4.0
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 RM TX16S

pfeerick, 2021-08-06:
Similar to #403

elecpower, 2021-08-06:
@pfeerick could be that bit shift in data conversion

pfeerick, 2021-08-07:
Yup... #403 was the issue I had in mind when you first mentioned that, and this now suggests it is not radio family specific...

elecpower, 2021-08-07:
Just checked the code and the 219 to 220 timer data conversion is not radio specific.
If you like I'll make a PR and then the theory can be checked?

elecpower, 2021-08-07:
> When a timer is set in OpenTX as On with a switch

In OpenTX setting to ON does not also allow setting a switch eg SAup as they are in the same list so I'm a bit confused.

Confirmed bug in the data conversion for some fields but ON is not one I can reproduce.

Also I have found a display issue on color radios where the timer counts down then appears to count up but really it is a negative value but not displayed with a negative sign. On B&W radios displays with negative sign and flashes.

@sohamsp14 what version of OpenTX are you converting from and is it the version that came with the radio or one you downloaded and installed? Also could you please attach an unconverted OpenTX file so I can try to reproduce what you are seeing.

elecpower, 2021-08-07:
@sohamsp14 do not worry about the OpenTX file as I have sorted it out. Thanks for reporting the issue

PR #572 (2021-08-14): Fix timer data conversion
Fixes #403
Fixes #566
closed 2021-08-14 by raphaelcoeffic