EdgeTX/edgetx #568: Invalid GPS coordinates

willannison, 2021-08-07:
Using ghost telemetry. Gps coordinates are fine on the osd and on the oled screen of the ghost module but what is in the telemetry page is not a valid GPS coordinate. 
[image]

pfeerick, 2021-08-08:
Interesting... the Latitude is spot on... but longitude is wrong... 

[image]

Maybe a signing error, since that's a negative longitude...

rotorman, 2021-08-09:
> Maybe a signing error, since that's a negative longitude...

Indeed, seems unsigned int is used:
https://github.com/EdgeTX/edgetx/blob/f5dec4e31edd8181e6a9925c31a03fa2cb659eef/radio/src/telemetry/ghost.cpp#L277-L278

https://github.com/EdgeTX/edgetx/blob/f5dec4e31edd8181e6a9925c31a03fa2cb659eef/radio/src/telemetry/ghost.cpp#L151-L157

raphaelcoeffic, 2021-08-09:
> Indeed, seems unsigned int is used:
>
> https://github.com/EdgeTX/edgetx/blob/f5dec4e31edd8181e6a9925c31a03fa2cb659eef/radio/src/telemetry/ghost.cpp#L277-L278 [...]

Make sense to use an unsigned to assemble bits together. The question is: is the conversion to the value used at the end for angle computation properly converted?

rotorman, 2021-08-09:
I guess it should be:
```
processGhostTelemetryValue(GHOST_ID_GPS_LAT, (int32_t)(getTelemetryValue_s32le(3) / 10));   
processGhostTelemetryValue(GHOST_ID_GPS_LONG, (int32_t)(getTelemetryValue_s32le(7) / 10)); 
```

as the processGhostTelemetryValue expects int32_t value field:
https://github.com/EdgeTX/edgetx/blob/f5dec4e31edd8181e6a9925c31a03fa2cb659eef/radio/src/telemetry/ghost.cpp#L99

Due to lack of hw, cannot test if that solves the issue though

raphaelcoeffic, 2021-08-16:
> I guess it should be:
> 
> ```
> processGhostTelemetryValue(GHOST_ID_GPS_LAT, (int32_t)(getTelemetryValue_s32le(3) / 10));   
> processGhostTelemetryValue(GHOST_ID_GPS_LONG, (int32_t)(getTelemetryValue_s32le(7) / 10)); 
> ```
> 
> as the processGhostTelemetryValue expects int32_t value field:
> 
> https://github.com/EdgeTX/edgetx/blob/f5dec4e31edd8181e6a9925c31a03fa2cb659eef/radio/src/telemetry/ghost.cpp#L99
> 
> Due to lack of Ghost hw, cannot test if that solves the issue though

I should be able to check, as I have a ghost module. However, no GPS. Maybe this can be simulated.

pfeerick, 2021-08-16:
If you have an Arduino handy, you might be able to mock it up...

This is the a dummy dataset for the TinyGPS library... if you just `Serial.println` each NMEA sentence and fed it in via serial connection that would probably do the trick. 

https://github.com/mikalhart/TinyGPSPlus/blob/cfb60e5e381c025ff7e52838897edab8a6e7d449/examples/BasicExample/BasicExample.ino#L9-L15

raphaelcoeffic, 2021-08-23:
In fact, it might be even easier to fix if I would have some sample telemetry packets showing the issue, so I could “unit test it”.

raphaelcoeffic, 2021-08-23:
> processGhostTelemetryValue(GHOST_ID_GPS_LAT, (int32_t)(getTelemetryValue_s32le(3) / 10));   
> processGhostTelemetryValue(GHOST_ID_GPS_LONG, (int32_t)(getTelemetryValue_s32le(7) / 10));

Possibly even this (conversion before division):
```
processGhostTelemetryValue(GHOST_ID_GPS_LAT, (int32_t)(getTelemetryValue_s32le(3)) / 10);   
processGhostTelemetryValue(GHOST_ID_GPS_LONG, (int32_t)(getTelemetryValue_s32le(7)) / 10); 
```

raphaelcoeffic, 2021-08-23:
@willannison would you mind testing #625 ? I think this should be it.

willannison, 2021-08-23:
Tested and it's working just fine! Thanks.
[image]

PR #625 (2021-08-24): Ghost GPS: convert to signed before dividing
Resolves #568
commit 8eab7bba (2021-08-24): Ghost GPS: convert to signed before dividing

Resolves #568
closed 2021-08-24 by raphaelcoeffic