EdgeTX/edgetx #569: NV14 does not boot

eshifri, 2021-08-07:
**Describe the bug**
NV14 does not boot with the current SW.

**To Reproduce**
Flash the FW.
Switch the TX ON.
Observe the screen giving a single flash and TX hangs.
After removing the batteries it  is possible to get into the bootloader screen.

**General information**
 - EdgeTX version current code
 - Hardware NV14

raphaelcoeffic, 2021-08-09:
@eshifri do you have an idea when it started not working? Maybe which commit did this?

eshifri, 2021-08-09:
@raphaelcoeffic 
I'm not sure. It was either change in the Cmake version or, more likely, unified ADC driver.
On the other hand I have spent a couple of hours on the ADC driver code and I had not found anything suspicious enough.

It was definitely not working after the CMake games, but I'm not sure I was building it after the ADC changes.

eshifri, 2021-08-10:
I rebuild the FW with -DUNEXPECTED_SHUTDOWN=NO and it works now.
Weird.

pfeerick, 2021-08-10:
I've noticed after the CMAKE update some options _seem_ to act differently at times... but it could just be my imagination or I hadn't noticed it before. Also, when you set a flag with CMAKE, you specifically need to turn it off again (unless you clean your build directory out)... i.e. explicitly setting `-DUNEXPECTED_SHUTDOWN=NO` on one build, and then not specifying it when re-running CMAKE means it will still be set to no... you actually have to set it to `YES` to make it toggle back.

raphaelcoeffic, 2021-08-10:
> I've noticed after the CMAKE update some options _seem_ to act differently at times... but it could just be my imagination or I hadn't noticed it before. Also, when you set a flag with CMAKE, you [...]

This is normal, the settings are cached in CMakeCache.txt.

pfeerick, 2021-08-10:
Yeah, the settings caching is normal... it was more things like what was happening with internalgps... not taking... or at least I think it was... it eventually seemed to do what it was told in the end...

eshifri, 2021-08-11:
Looks like it fixed itself. At least i do not see this problem any more.

closed 2021-08-11 by eshifri