EdgeTX/edgetx #570: An “alternative” font

timemaster5, 2021-08-07:
Hi guys, I wanted to try EdgeTx and realised that there are no build options like we used to have in OpenTx. I assume everything is enabled for now, which is perfectly fine. I am only wondering about that alternative font, I know its probably a lowest priority, but for me on a small 128px display, it is for some reason much more readable font than the original one. Can I somehow enable it?

pfeerick, 2021-08-08:
Is that the sqt5 font? I think it can be for the B&W transmitters... What one are you using... T-Lite? I can do a custom build for that if you want it...

timemaster5, 2021-08-08:
Yes, it is. Sorry, I could swear, I saw it somewhere called "alternative".
Never mind. If you could do that, it would be awesome.

pfeerick, 2021-08-09:
Yeah, when you hover over the option in opentx companion it says it's the 'alternative sqt5 font', so that is probably what you are thinking of ;)

I assumed the following for the build settings (looks like S and Pro are the same build)... let me know if you need/want something different:

```
-DPCB=XLITES -DHELI=NO -DGVARS=YES -DLUA=YES -DFONT=SQT5
 ```

[edgetx-f5dec4e-xlites-noheli-gvars-lua-sqt5font.zip](https://github.com/EdgeTX/edgetx/files/6951680/edgetx-f5dec4e-xlites-noheli-gvars-lua-sqt5font.zip)

timemaster5, 2021-08-09:
Thank you very much, I checked my settings, and OpenTX reporting these: Bluetooth, crossfire, ghost, frexr9m, noheli, internallaccess, multimodule, lua, luac. Currently, I use all of it with different models. Maybe except Bluetooth which is only for a trainer mode in my case.

pfeerick, 2021-08-10:
Man... was the sqt5 font not enough for you? 😆 

`simu` is giving me this, so it should be everything you're after...
[image]

```
cmake -Wno-dev -DPCB=XLITES -DHELI=NO -DLUA=YES -DGVARS=YES -DFONT=SQT5 -DMODULE_PROTOCOL_FLEX=YES -DAFHDS3=NO ../
```

[edgetx-f5dec4e-xlites-noheli-lua-gvars-sqt5font-flexr9m-noafhd3s.zip](https://github.com/EdgeTX/edgetx/files/6958275/edgetx-f5dec4e-xlites-noheli-lua-gvars-sqt5font-flexr9m-noafhd3s.zip)

timemaster5, 2021-08-10:
Haha, it's well-used radio I guess :) Thank you very much, I will test and try to be more useful :)

pfeerick, 2021-08-11:
No problem, hope it works for you!

closed 2021-08-11 by pfeerick

timemaster5, 2021-08-11:
Just tested, and works perfectly!