EdgeTX/edgetx #571: Timers Not Incrementing In Synchronized Manner

unklstewy, 2021-08-07:
While configuring a new quad model I was setting up some timers on my main screen when I noticed that the timers appear to be out of sync.

Timer2 Lagging:
[Youtube: Timers Out of Sync](https://youtu.be/z5PVKw-6qhk) 

Setup:
Radio: Radio Master TX16S Max
EdgeTX: 2.4.0

TIMER1:
[image]

TMER2:
[image]

TIMER1:
Starts with First Arm (SF FWD) and Pauses with Disarm (SF BACK). Reset Configured with:

Logical Switch:  L02 
[image]
&
Special Function: SF2
[image]

TIMER2: Stars with Arm Switch (SF FWD) and Stops+Resets with Special Function: SF1
[image]

Are these timers running sequential interrupts, or is the use of special functions and logical switches causing TIMER2 to have a delay in the GUI?

-Casey

raphaelcoeffic, 2021-08-09:
@unklstewy I think this might be caused by the widget refresh on screen not being synced. They refresh with the same frequency, but not at the same moment.

unklstewy, 2021-08-10:
I imagine that be something in LUA's sequential processing.  I guess I wasn't expecting a 300ms -500ms delta in the display times.

raphaelcoeffic, 2021-08-14:
> I imagine that be something in LUA's sequential processing. I guess I wasn't expecting a 300ms -500ms delta in the display times.

In this case it's not LUA, it's one of the C++ Widgets. We might try to refresh whenever the value has changed instead of some fixed (non-synced) pace.

unklstewy, 2021-08-15:
@raphaelcoeffic 

Thank you for taking the time to look into it and get back with me.  You guys are killing it with the development pace!  Kudos.

gagarinlg, 2023-01-11:
This does not happen in main

closed 2023-01-11 by gagarinlg