EdgeTX/edgetx #574: Radiomaster TX16 - Internal Bluetooth

IdefixRC, 2021-08-08:
Hi there,

OpenTX provides a version of the firmware which enables the TX-16 internal Bluetooth port.
I could not find any info in the documentation regarding BT for TX16 so I assume it is not included in the EdgeTX FW.
Would it be possible to release a version of the EdgeTX firmware with BT enabled?

Thanks !

pfeerick, 2021-08-09:
When we have the online build system in place, you'll be able to pick options like this for yourself (this is what is happening when you hit the download option in OpenTX companion - it's looking to see if there is a cached build already of the firmware you've requested, or it will task their build server to build it on demand). 

I've assumed these build options, and built off the latest commit firmware - let me know if you want a different configuration or version:

``` -DBLUETOOTH=Y -DGVARS=Y -DHELI=N -DLUA=Y```
[edgetx-tx16s-f5dec4e-bluetooth-gvars-noheli-lua.zip](https://github.com/EdgeTX/edgetx/files/6951797/edgetx-tx16s-f5dec4e-bluetooth-gvars-noheli-lua.zip)

IdefixRC, 2021-08-09:
Thanks a lot. Could you drop one with the same settings but heli enabled
please? Thanks a million

pfeerick, 2021-08-09:
lol... What... ... you actually *use* heli? grrr! It'll cost you! 😝 

[edgetx-tx16s-f5dec4e-bluetooth-gvars-heli-lua.zip](https://github.com/EdgeTX/edgetx/files/6951894/edgetx-tx16s-f5dec4e-bluetooth-gvars-heli-lua.zip)

IdefixRC, 2021-08-09:
Haha, thanks!

pfeerick, 2021-08-09:
No problem. Let me know if you need anything else ;)

closed 2021-08-09 by pfeerick

SilentZen, 2021-08-10:
Tested this build and compiled it myself. Bluetooth discover seems to be missing under model/trainer on both.

pfeerick, 2021-08-11:
That for that @SilentZen ... we'll track it in https://github.com/EdgeTX/edgetx/issues/583

Millerboy3, 2021-09-01:
Tested your custom build and it works like a charm to show things in menu's on my tx16s. Good work.  Will be awesome when the build server is up and we can enable it that way.   Thanks for this!

SilentZen, 2021-09-02:
Where is the source for this fix?

pfeerick, 2021-09-02:
? No fix, Millerboy3 said it works *sans discover* (minus discover) ;) Nothing has changed here yet.

SilentZen, 2021-09-02:
Ahh, thanks