EdgeTX/edgetx #576: CRSF as internal protocol

NemesisXB, 2021-08-09:
Feature request. Lots of guys over on the expresslrs discord is asking for CRSF as an internal module protocol.

JyeSmith, 2021-08-09:
Im down for this.  It opens up the way for more internal module development, full duplex, and no pesky inverters.

pfeerick, 2021-08-10:
As may or may not have been mentioned on the Discord, this is certainly going to happen - basically why should the internal RF option be any different to the external - just because a radio was manufactured with X RF capability/module it doesn't mean it will always have it once you have us tinkerer's messing around in the guts of it ;)

raphaelcoeffic, 2021-08-10:
The idea was the following: basic settings (protocol) for the internal module will be in “Radio Settings”, as you normally don’t exchange these modules every day.

FractalEngineer, 2021-08-31:
Very much this, as well as the internal multiprotocol option; would open up the opportunity to replace the stock RF module by whatever you use the most (looking at you Frsky) 

>Don't Jumpers/Radiomaster and other internal multiprotocol radios work like this already?