EdgeTX/edgetx #577: Immersion RC Ghost

SPCORRIV, 2021-08-09:
Edge 2.4

Ghost Lau script seems to be there but on opentx you could go to system down to ghost and setup a reciever IE Bind change ghost protocal. On Edge I have to use Module on back of transmitter to do set up. Would be nice to have the ghost screen back on the transmitter scree.

Radio TS16 MAX
ImersionRC Ghost Module

Thank You

pfeerick, 2021-08-10:
I don't have Ghost hardware, so I'm not that familiar with it... but do you need to set the protocol in model settings to "GHST" for it to work? I see from [this video](https://youtu.be/VavnWYUSb-E?t=94) there should be another menu entry at the bottom titled "Ghost Menu", which judging from where it is in the list is provided by the radio firmware, not a lua script. From a quick look around the code, it should be provided... ah, wait... it looks like it has been explicitly disabled for now... https://github.com/EdgeTX/edgetx/commit/8cc5e7777561cea16df227579a9ae6d5a34e6714 ... 

@raphaelcoeffic I'm presuming something was broken?

raphaelcoeffic, 2021-08-10:
> I don't have Ghost hardware, so I'm not that familiar with it... but do you need to set the protocol in model settings to "GHST" for it to work? I see from [this video](https://youtu.be/VavnWYUSb-E?t=94) there should be another menu entry at the bottom titled "Ghost Menu", which judging from where it is in the list is provided by the radio firmware, not a lua script. From a quick look around the code, it should be provided... ah, wait... it looks like it has been explicitly disabled for now... [8cc5e77](https://github.com/EdgeTX/edgetx/commit/8cc5e7777561cea16df227579a9ae6d5a34e6714) ...
> 
> @raphaelcoeffic I'm presuming something was broken?

The “Ghost” menu has been disabled shortly before 2.4.0, as I did not have the time to test it properly. On my first test, it was just displaying some random bits, so that I had to disable it.

tonycake, 2021-11-10:
So just removing the #ifdef 0's the menu is fully functional, and looks pretty good. 
The only thing I found that is not ideal is that the return functionality backs all the way out of the menu instead of just back one level. 
I will try to fix that, but I see no reason why the #ifdef 0 can't be removed in the mean time.

tonycake, 2021-11-10:
Not sure I understand what is going on with the return behavior. 
The code from OpenTx and EdgeTx is basically the same within `RadioGhostModuleConfig::onEvent() `

Wondering if something outside of this module is taking the single return key hit as a request to leave the menu completely. 

Anyone have a good understanding of what changed in this area between OpenTx and EdgeTx?

raphaelcoeffic, 2021-11-10:
@tonycake do you think the same would be possible with LUA? It would be in fact much better. Additionally, it would be cool to have a widget, especially with the new touch support and full screen mode.

tonycake, 2021-11-11:
Possibly... personally I am no Lua expert, so I would need the help of one. 
On the touch side, for the new Orqa Ghost radio, we have modified the API to the Ghost module so that it is more conducive to a touch interface. 
The built-in menu is a lot more responsive than Lua equivalents (at least when we looked at it last), so it would be a shame to loose that side.

tonycake, 2021-11-11:
If Lua is indeed the chosen path, it might be a good idea to enable the current Ghost menu (even with the non-optimal return key behavior) so that the many Ghost pilots that have switched to EdgeTx can use the front-side menus. @raphaelcoeffic are you ok with me doing this?

raphaelcoeffic, 2021-11-11:
@tonycake ok, here is the plan:
- re-enable the built-in tool as a transition solution.
- move to LUA widget (incl. fullscreen support for config)
- and solve whatever responsiveness issues there are on the way.

there are a couple good reasons for doing so IMHO:
- LUA widgets offer so much more
- I don’t want to make it look like there are “special treatments” for certain vendors (TBS and ELRS do not enjoy the chance of a built-in tool)

FYI, we have already moved to removing external LUA tools from the SD card distribution, so that users are not bothered without outdated tools like the old crossfire.lua. We want to be vendor neutral. Having the FrSky “special treatment” as a legacy is already enough.

Plus, you have the possibility to be much more agile, basically not depending on us releasing anything when you need a change. EdgeTX should focus on operating system features and offering the best possible runtime environment for module vendors of all sorts (for example with good and fast LUA widget API).

tonycake, 2021-12-15:
So it seems that the remaining return key behavior issue is related to an error in `RadioGhostModuleConfig::onEvent`

In the  `case EVT_KEY_BREAK(KEY_EXIT):` case, `Page::onEvent(event);` should not be called. 

 `Page::onEvent(event);` kills the page for both the `EVT_KEY_BREAK(KEY_EXIT)`, and `EVT_KEY_LONG(KEY_EXIT)` cases.

pfeerick, 2021-12-16:
So does this work then?

```diff
diff --git a/radio/src/gui/colorlcd/radio_ghost_module_config.cpp b/radio/src/gui/colorlcd/radio_ghost_module_config.cpp
index f5a835452..d8e823077 100644
--- a/radio/src/gui/colorlcd/radio_ghost_module_config.cpp
+++ b/radio/src/gui/colorlcd/radio_ghost_module_config.cpp
@@ -137,7 +137,9 @@ void RadioGhostModuleConfig::onEvent(event_t event)
       deleteLater();
       break;
   }
-  Page::onEvent(event);
+  if (event != EVT_KEY_BREAK(KEY_EXIT)) {
+    Page::onEvent(event);
+  }
 }
 #endif
 ```

tonycake, 2021-12-16:
Unfortunately not... well it does help significantly, but there is still a case that I don't understand. 

When Ghost backs out of the root menu, it sends a GHST_MENU_STATUS_CLOSING status back to the radio. This is used to close the Ghost menu. 
I can process this in RadioGhostModuleConfig::onEvent, but I don't know how onEvent gets triggered by the telemetry event. 

I think I need to pick 3djc's brains, he was the original author.

tonycake, 2021-12-16:
This fixes it: https://github.com/EdgeTX/edgetx/pull/1261

closed 2021-12-16 by pfeerick