EdgeTX/edgetx #579: [Bug] [NV14] can't exit out of widgets setup

LimitlessGreen, 2021-08-10:
Hello,
I'm not able to exit out the *Setup widgets* dialog. This behavior effects only the first tab where the top elements are changeable. Also only the first two boxes should have the ability to put a widget in (otherwise bat, time and RSSI widgets will be overdrawn). 

**To Reproduce**
Touch the screen -> Screen Settings -> First Tab (User Interface) -> Setup widgets

**Expected behavior**
Touch the logo on the top left to exit.

**In short**
- [ ] Make the exit at "Setup widgets" work
- [ ] Hide the right two widget boxes  

**General information**
 - EdgeTX version (2.5.0 nightly, #533)
 - Hardware (FlySky Nirvane NV14)
 
 ---
PS: Are the not selectable themes and the only white background a symptom of "not yet implemented"? :octocat: 

_Thanks of the nice project :)
-- Limitless Green --_

LimitlessGreen, 2021-08-10:
It's possible to exit out with the exit button, just not with the logo.

eshifri, 2021-08-10:
I can reproduce it. Will take a look.

raphaelcoeffic, 2021-08-14:
Fixed by #584

closed 2021-08-14 by raphaelcoeffic