EdgeTX/edgetx #581: [Bug] [NV14] SG & SH interchanged

LimitlessGreen, 2021-08-10:
If you try to change SH, the radio is detecting a change in SG and vice versa.
This caught my attention because I was trying to make SG an FM switch (3 states) but the radio was detecting a toggle switch. 

**Describe the bug**
SG & SH interchanged

**Screenshots**
[image]

**General information**
 - EdgeTX version (2.5.0 nightly, #533)
 - Hardware (FlySky Nirvane NV14)

---
_Thanks for the nice project :)
-- Limitless Green --_

eshifri, 2021-08-10:
Yes, this is a new problem - this thing used to work correctly.

Since you are looking at the switches :-), can you please take a look at the list of options, available for logical switches?
I think !SA is missing on NV14.

LimitlessGreen, 2021-08-10:
This one?

[image]
[image]

eshifri, 2021-08-10:
No, sorry I did not make myself clear: if you go to AND switch SA up/down/neutral will be there, but no sign of !SAu, !SAd

LimitlessGreen, 2021-08-10:
Sorry, I'm a bit new to this stuff. :sweat_smile:
Yes SA is there, but what is !SAu?
I'm assuming an inverted something? :sweat_smile:

eshifri, 2021-08-10:
Yes :-), inverted SA up or down

LimitlessGreen, 2021-08-10:
Ahh yes, you're right!
The inverted ones are beginning with !SB
[image]

eshifri, 2021-08-10:
Actually, this is correct: does not make sense for two-position switches.

pfeerick, 2021-08-11:
It does, and it doesn't... I know myself I was confused at first when I didn't see them on the T-Lite, then realised they were the two position switches, and there are arguments for them to be omitted (since a two position switch is only up or down, there isn't that 'indeterminate' middle position) and for them (because you want to simply say not up, rather than down, well, because). That is why https://github.com/EdgeTX/edgetx/issues/178 is still open for discussion as to whether it should stay as it is or change.

raphaelcoeffic, 2021-08-14:
Fixed by #584

closed 2021-08-14 by raphaelcoeffic