EdgeTX/edgetx #582: [BUG] [NV14] wrong display color for BMPs

LimitlessGreen, 2021-08-10:
The review of an BMP file in the NV14 has the wrong color.
I also tested JPG and PNG and these are fine. 

**Screenshots**
[image]
[image]

For me, it looks like the NV14 is displaying BRG instead of RGB.

**General information**
 - EdgeTX version (2.5.0 nightly, #533)
 - Hardware (FlySky Nirvane NV14)

LimitlessGreen, 2021-08-10:
I uploaded the sample bmp. Maybe jpg wasn't ideal for this case.
The color top bar from picture to picture in picture from the original: 
- #105c98 rgba(16,92,152,255)
- #d05070 rgba(208,80,112,255)
- #c8c080 rgba(200,192,128,255)

pfeerick, 2021-08-11:
Actually, it's not NV14 specific, it is the same on the TX16s, and was something that changed a few months ago, so is possibly something to do with libopenui... https://github.com/EdgeTX/edgetx/issues/293

closed 2022-09-18 by pfeerick