EdgeTX/edgetx #583: Bluetooh

Guillesan, 2021-08-10:
Hello guys.  I have installed Edge 2.4 in a Jumper T16, I have a bluetooh module installed in it, with OpenTX it worked for me but in Edge there are no traces of being able to configure it or see the connection.  my question is whether the bluetooh is implemented or not.  Thanks

SilentZen, 2021-08-10:
Trainer mode under model when selecting Master/Bluetooth seems to be missing Discover and instead has channel range options.
No option to select dist address.

pfeerick, 2021-08-11:
Bluetooth is a compile time option, so will not be present unless you build it yourself (or get someone to build it for you) until we have the online build system up and running. And as has been pointed out by SilentZen, it seems like some options are not implemented / missing still.

iggut, 2021-08-17:
NV14 module is also not used by the looks of it.

closed 2021-09-13 by pfeerick