EdgeTX/edgetx #585: Audio starts playing before applying volume as set by pot

NemesisXB, 2021-08-12:
**Describe the bug**
When audio volume is assigned to a pot and the volume is tuned using the pot, on startup it still plays a short bit of the welcome audio clip at the max volume before turning the volume to the "set" value using the pot. It is worst when the pot is turned all the down (volume off), then it will play a short bit of the audio clip before turning it off.

**To Reproduce**
Create a special function and assign a pot to control the volume. Turn the pot all the way down. Power off the radio, power on the radio. Ensure that there is a valid welcome audio clip.

**Expected behavior**
The audio volume should always get the value from the set pot before playing the audio

**General information**
 - This is the same in OpenTX, so it is not an EdgeTX specific issue
 - RM TX16S

pfeerick, 2021-08-12:
I thought there was an issue for this already, but it seems not. There was some discussion on this before, with the idea that both brightness and volume being set via pot should actually be part of radio settings, rather than a global function, thus making it possible to use them earlier in boot. This would fix up issues like this, and also probably ones like https://github.com/EdgeTX/edgetx/issues/53

raphaelcoeffic, 2023-07-08:
Maybe I should finally finish #2256 and add volume control as discussed at the time.