EdgeTX/edgetx #589: USB mode option in Model Settings to override Radio Settings

jogybaer0815, 2021-08-14:
TX16
It is already a nice thing to answer the USB mode by touch when I connect the transmitter to the computer.
An option in the model settings would be very elegant to set a default according to the model (ask, joystick, storage, serial) and thus to override the SYS radio setup.
Then, for example, the selection would be omitted if I connect the transmitter to the USB port in order to practice on the simulator, because I use my own model memory for the simulator.
Other models require a different choice e.g. storage ...
Thanks for this great system

pfeerick, 2021-08-14:
That could work quite nicely - adding setting like "Model" in the SYS/RADIO Setup tab USB mode dropdown, which would default to Ask with new models, thus allowing either per model (with the default setting preserving current default) settings or a global value.

gagarinlg, 2023-05-05:
This sounds interesting.
I would prefer to have it like ADC and other settings, where you can set in the model if you want to use the global setting or override it

closed 2026-10-30 by pfeerick