EdgeTX/edgetx #591: Custom Scripts don't see Lua outputs name

BartSoj, 2021-08-14:
Simple Lua program but the output name doesn't show
[image]
`local inputs = {
                 { "Input", SOURCE },
                 { "Percent", VALUE, -100, 100, 5 }
               }

local outputs = { "GtSt" }

local function run(i, p)
  return i*p/100
end

return { input=inputs, output=outputs, run=run }`

raphaelcoeffic, 2021-08-16:
@BartSoj could you please state which radio & firmware version you are using?

BartSoj, 2021-08-16:
I'm using Radiomaster tx16s with Edgetx v2.4.0

raphaelcoeffic, 2021-08-16:
> I'm using Radiomaster tx16s with Edgetx v2.4.0 ([09a6580](https://github.com/EdgeTX/edgetx/commit/09a65807ecee1b8b4b583b795931f8bc6690d437))

Thx, could you please re-try with latest nightly? I think we had already fixed something like that.

BartSoj, 2021-08-16:
tried but didn't help

pfeerick, 2021-08-16:
This is the PR you were thinking off that *should* have fixed this, but it seems to have gone bonkers (sorry, technical term 😆) since... https://github.com/EdgeTX/edgetx/pull/436

TX16S, using the nightly from probably two days ago, adding the mix above as `Mix1.lua`, and the example from the PR as `Mix2.lua`, strange stuff happens. Regardless of which mix I put in the first line, it seems to error... 

[image]
[image]

But you can quite clearly see the second 'instance' of `Mix1` did something... 🤪 

And well, the mixes pages themselves were a bit bare... :-O
[image]
[image]

I've never used custom mixes, so could have buggered something up... I just copied the text into a .lua text file, and shoved it into a text file with .lua extension `/SCRIPTS/MIXES` with a less then six character name (hence Mix1, Mix2)... reloaded the model just to be sure... no change. 

As a side issue, the 'Reset' entry in the context menu for the custom scripts page doesn't seem to work.

raphaelcoeffic, 2021-08-16:
Hmm… seems like the feature needs some attention to keep working 😇

pfeerick, 2021-10-02:
Is there any chance you could look into this @jfrickmann? It looks like somewhere between 

https://github.com/EdgeTX/edgetx/commit/762ce4ce1fa725b8539eb93a4af4baa36f0a81bf
and 
https://github.com/EdgeTX/edgetx/commit/b0732e156406017e30b6f20cd19dd02034996a62

this broke, and it seems related to the pre-emption PR. `simu`, as well as radio, so that should be of some help. 

The test script I was using was https://raw.githubusercontent.com/RCdiy/OpenTXLua/master/MIXES/GtStd.lua

jfrickmann, 2021-10-02:
I will look into it

jfrickmann, 2021-10-02:
I copied the script and fired up the 2.5rc2 simu. It looks fine here - am I missing something?
I used aile and elev as inputs.
[image]
[image]

pfeerick, 2021-10-03:
Right, so with `cmake -DPCB=X10 -DPCBREV=TX16S -DLUA=Y ..` option (i.e. TX16S, with LUA on for mixer scripts), this is what I get when I add the mixer script linked above on the custom screens page..
[image]

and it shows this back at the main screen

[image]

but if I added it to a second lua line, I get the following

[image]

However, if I revert back to commit 762ce4c, it is fine. 
[image]

[image]

Am I missing something about mixer scripts, or perhaps it's specific to the radio type?

jfrickmann, 2021-10-03:
I dont know - I was just using TX16S simu from 2.5rc2 Companion. Can you look in the Lua log and see why that error happens?

pfeerick, 2021-10-03:
Ah ha, well, therein lies the problem... I have now tried with the Companion 2.5 simulator, and that is working fine... so 1) Companion simulator is not reflecting radio hardware behaviour in this instance and 2) Companion simulator is not behaving the same as the same commit `simu`, which does match hardware behaviour. 😱 

This is the log from `simu` [simu_591.txt](https://github.com/EdgeTX/edgetx/files/7272697/simu_591.txt)

And the log from `simulator` from companion [simulator.txt](https://github.com/EdgeTX/edgetx/files/7272698/simulator.txt)

I do see there is a line `-E- : attempt to call a nil value` - but it also appears much earlier on, before the mixer script is even enabled, and will also appear even if the file isn't present. It seemed unrelated, but I also note that message isn't present in the earlier 762ce4c repo state of simu/firmware. [simu_591_oldcommit.txt](https://github.com/EdgeTX/edgetx/files/7272700/simu_591_oldcommit.txt)

Note in all cases I'm doing mostly the same thing - nuking the simulator radio and model settings, starting it afresh, closing it, and then starting again, going to custom scripts tab, picking the mixer script, go out of the page, go back in, no option are shown / options are shown, and then close it all down. I may also try giving the instance a name, but that didn't appear to make any difference.

jfrickmann, 2021-10-03:
Yeah, the nil value has been there for a while, maybe I should figure out where it comes from 😉 
I will build simulator2_6_ under Linux and see what happens.

jfrickmann, 2021-10-03:
I just fired up Companion for X9D+, and it does not give me "-E- attempt to call nil value", so that one must be widget related.

jfrickmann, 2021-10-03:
So I just fired up TX16S under simulator26, freshly built from the current main (0cbb14bc1), and it still works jus fine ...
Now I will grab my radio and see how it looks under 2.5rc2

jfrickmann, 2021-10-03:
On my TX16S w/ the FW from 2.5rc2 (994c4ff) it errors, but firmware built with the current main head (0cbb14bc1) seems to work just fine. I would say that something is rotten with 2.5rc2, or the build system used for building RCs...
Another thing I noticed, is that it always says 0% on the right side in CUSTOM SCRIPTS - should it give the current value of the first output or something?

pfeerick, 2021-10-05:
The build system for RCs is the same one that is building every other build - so it isn't that. 

It is possible there is something different with the 2.5rc2 firmware, but 0cbb14b is no different for me when I try the [version from the github build](https://github.com/EdgeTX/edgetx/runs/3779268254) , so I don't know what is going on here. If you are were comparing a self-build 0cbb14b, it sounds like there is something different in your build environment which is letting it work, which is different to both the github actions build, and mine (which is using the same docker container used by the github actions).

closed 2021-10-05 by pfeerick

jfrickmann, 2021-10-05:
I just thought of something. The LUA compile flag is a bit odd: YES=Full Lua support incl. mixer scripts; ON=Lua support but no mixer scripts; OFF=no Lua support.
In OTX, the default was ON, and that's why I stopped using it, because I had to explain to my users how to set it to YES and download and install FW again. I bet the same is the case for our standard builds? If that is the case, then we probably build the Mixer script GUI, but there is no support for it under the hood.
So we should probably:
- [ ]  change gui/colorlcd/CmakeLists.txt to only include model_mixer_scripts.cpp for LUA=YES
- [ ] Consider setting the default value -DLUA=YES

I will test it when I get a moment, and let you know what I found, with a PR if it worked.

pfeerick, 2021-10-05:
I got distracted, so you beat me to this. Yes, that was the problem... and it confused me because the tab was visible...

https://github.com/EdgeTX/edgetx/blob/f970ab1e661fcc90fa2bcb15d35ee323b81621ac/radio/src/gui/colorlcd/menu_model.cpp#L52-L54

This could also be made to test for YES rather than just ON, as it is obviously incorrect atm 🤦 

IMO, it being linked to LUA flag is completely incorrect - it should have had it's own flag, since `LUA` is not `LUA_MIXER_SCRIPT` 🤮

jfrickmann, 2021-10-06:
In radio/src/CmakeLists.txt:
```
  if(LUA STREQUAL YES)
    add_definitions(-DLUA_MODEL_SCRIPTS)
  endif()
```
So we just need to apply that to compiling and adding the ModelMixerScriptsPage.

pfeerick, 2021-10-06:
Nice... and then can also forget about LUA=YES, since LUA=ON is generally on by default, and LUA_MODEL_SCRIPTS will probably do the model/mixer scripts whereever it is needed...

jfrickmann, 2021-10-06:
PR #869 on the way...

jfrickmann, 2021-10-06:
So I learned the the defines only work in the source code, not in CMakeLists.txt. So I had to use if(LUA STREQUAL YES) in gui/colorlcd/CMakeLists.txt to get the mixer script menu compiled.