EdgeTX/edgetx #592: No timer suspend after start

sfjuocekr, 2021-08-14:
I use the arm switch to start a timer, on OpenTX these timers would pause when the switch was not set and run when the switch was set.

EdgeTX behaviour currently only starts timers and there is no way to pause a timer.

eshifri, 2021-08-15:
Works for me in the current SW.

pfeerick, 2021-08-15:
What Transmitter, and what settings are you using? For instance, on a recent nightly on the TX16S, `ON` and SFdn makes the timer only be 'running' when SF is in the down position. It runs when it is down, and stops when it is up.

raphaelcoeffic, 2021-08-15:
The timer is probably set to “Start”, which is exactly what the function does: it starts the timer the first time. You should use “On” instead, together with your arm switch.

MervDale, 2021-08-15:
Hi, it works perfectly well when coupled to a switch, in this case when the motor is armed the timer will stop and start with the throttle.

cheers
[image]

pfeerick, 2021-08-17:
@sfjuocekr I've just checked for Timers 1 and 2, if you're using any trigger other than Start, the timer starts when you set your switch, and stop when you unset it. So I'm going to close this issue for now, unless you have something more to add.

closed 2021-08-18 by pfeerick

sfjuocekr, 2021-08-22:
Ah yea, I got it working and forgot to close this!

I think what happened is that I imported my models and it was set to "START" for some reason, "ON" indeed has the expected behavior.

raphaelcoeffic, 2021-08-22:
> Ah yea, I got it working and forgot to close this!
> 
> I think what happened is that I imported my models and it was set to "START" for some reason, "ON" indeed has the expected behavior.
> 
> I was also looking for a function to stop a timer to put on a switch, but you can only set a timer to some value?

That was a conversion which has been fixed since in `main`.

sfjuocekr, 2021-09-01:
Ok, go to model > special functions > add a function to a switch.

How do I start and stop a timer from here?

I can set and reset them, but not other functions like start or stop?

pfeerick, 2021-09-01:
Because there is no need! 

When you set the mode to 'ON', and set a switch, like here... the timer will only run when the switch is the indicated position. Otherwise it will stop. Same for Throttle, Throttle Start, etc. "Start" is the only unique one. 
[image]

sfjuocekr, 2021-09-08:
Well I was thinking you might want to stop the timer once a certain condition is met regardless of the switch setting, like a failsafe or specific GPS speed?

I know it works that way from the timer setup on the model page.

pfeerick, 2021-09-09:
Ok, so now we're going onto a different topic to the initial issue - i.e. a feature request for something that doesn't exist on OTX at all yet. As special functions exist to set or reset the timer, but there are no special function on OTX or ETX to either pause or resume a timer.