EdgeTX/edgetx #605: The List of Multi-Protocol-Numbers is out of sync to the numbers used in the MPM itself

wimalopaan, 2021-08-19:
The enum of protocol numbers here

https://github.com/EdgeTX/edgetx/blob/b7a10d32bd9b5feb2f83a77efe015a05caa21905/radio/src/pulses/modules_constants.h#L104

is out of sync to the list here

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/324419241f73a24ce3b13c79896238ceb412cafa/Multiprotocol/Multiprotocol.h#L29

Additionally, because the names of proto/subproto the module sends back are ignored (as opposed to OpenTx), this is not clear to the user.

Additionally the list stops at the Config-Protocol, so the newer ones of the MPM are not shown.

pfeerick, 2021-08-20:
I'm thinking that this and https://github.com/EdgeTX/edgetx/issues/159 should be worked on together... #159 will probably get closed through https://github.com/EdgeTX/edgetx/pull/580 but that more a temporary fix. There is probably enough work needing to be done around the MPM that it deserves it's own project to organise what needs to be done.

kevinkoenig, 2021-08-20:
I created pull #580 and I would be willing to help solve this problem in a more generic way.  Let me know how I can help.

closed 2021-09-14 by pfeerick