EdgeTX/edgetx #609: On NV14 exit button does not work in the new topbar layout

eshifri, 2021-08-20:
**Describe the bug**
The top left icon in the top bar does not work as "Exit" button.
This happens in all screens: Model Settings, Radio Settings, Channel Monitor, Screen Settings

**To Reproduce**
Touch EdgeTX icon in the top left corner of the screen.
Select Model Setup in the carousel (or any other option).
Touch top left icon in the topbar.
Nothing happens.

**Expected behavior**
Top left button should work as exit.

**General information**
 - EdgeTX latest code
 - Hardware NV14.

eshifri, 2021-08-21:
Actually, it does not happen all the time.
It works after reboot.
Than problems appear.
Sliding top-bar back and force restores functionality for some time.

richardclli, 2021-08-23:
Maybe the button just too small for big fingers.  I never have this problem, well maybe my finger is small.

eshifri, 2021-08-23:
My fingers are also small and it did happen before the changes to top bar.
So while it very possible that the problem is the small button, looks like something has changed.

pfeerick, 2021-08-25:
Got this a few times last night - I do have to wonder if it is the size of the touch area, as the edges are really hard to get at with the bezels... have to try with a touch pen to see if that is all that it is.

eshifri, 2021-08-25:
I think you are right: I have tried to use a touch-pen (does not work too well), but i realized I need to push a pen/ a finger into the very corner of the screen.

richardclli, 2021-09-02:
Not sure how the touch control is implemented.  For best experience, the touch control can use some fuzzy algorithm to guess the user's intention and fire appropriate event.  i.e. Even the touch point is outside the display area of a button, we can still trigger the button based of distance between the the touch point and different touch triggerable widgets.

I guess the current implementation is the touch triggerable area = display area.

closed 2021-09-05 by eshifri

pfeerick, 2021-09-28:
@eshifri If you're interested, I found out the *what* is happening when playing with simu... the hitbox is only about this big 😆 ... which is the *why* as to why it's so awkward on the NV14 (because of the screen bezel). Now, as the what bit of code controls that, I can't place that yet - whatever it is, it seems the same on model settings and radio settings, but the full logo/icon elsewhere...

[image]