EdgeTX/edgetx #610: NV14: Selecting CRSF from external drop down causes the radio to lock up

petervieth, 2021-08-20:
**Describe the bug**
Selecting CRSF from external drop down causes the radio to lock up

**To Reproduce**
Turn radio on
Model Setup
Scroll down to external
Open drop down
Select CRSF
My radio displays red/blue junk and then screen goes black. Pulled the batteries to reset.  
I was able to repeat this 3 times.

**Expected behavior**
CRSF selected

**General information**
Used nightly build downloaded 3 hours ago
FlySky NV14

raphaelcoeffic, 2021-08-23:
@petervieth did you try with latest nightly? Or which version did you test with? I tested yesterday but did not experience the issue.

petervieth, 2021-08-23:
Unfortunately I'm over a decade out of software programming and not real hip on using github etc.  My download's filename includes b7a10d32 if that's any use.  I downloaded it about 8 or 9am Pacific time on the date of my submission.  Brand new NV14 pulled out of the box the same morning.  Poked around a bit and everything looked good but anytime I enabled CRSF for model01 the screen went mostly black but with blue and red dots on it which faded out. Required a battery pull to restart as the power buttons didn't work.    I have a Crossfire Micro TX from about a year ago that I did update (both TX firmware to 4.11 and wifi module to 1.17) and which works in my TX16s.

raphaelcoeffic, 2021-08-23:
> Unfortunately I'm over a decade out of software programming and not real hip on using github etc. My download's filename includes [b7a10d3](https://github.com/EdgeTX/edgetx/commit/b7a10d32bd9b5feb2f83a77efe015a05caa21905) if that's any use. I downloaded it about 8 or 9am Pacific time on the date of my submission. Brand new NV14 pulled out of the box the same morning. Poked around a bit and everything looked good but anytime I enabled CRSF for model01 the screen went mostly black but with blue and red dots on it which faded out. Required a battery pull to restart as the power buttons didn't work. I have a Crossfire Micro TX from about a year ago that I did update (both TX firmware to 4.11 and wifi module to 1.17) and which works in my TX16s.

This might sound like a stupid question: did you try with the module plugged in, or without a module plugged as well? This might be caused by the power regulation for the external module. You could also try it out with USB cable plugged, and USB set to joystick or serial. The radio is then powered partly via USB, which makes it much more stable.

petervieth, 2021-08-25:
Only with the module plugged in and no USB power.  I kinda assumed if it works with the FlySky firmware, it would work with this one?  Flew five packs on the FlySky today no issues.

raphaelcoeffic, 2021-08-25:
> Only with the module plugged in and no USB power. I kinda assumed if it works with the FlySky firmware, it would work with this one? Flew five packs on the FlySky today no issues.

It should, but also depends on the charging state. I only have these kind of issues when I plug a module while the radio is running and the module is turned on while plugging. This is a real hint at power (hardware) issues.

richardclli, 2021-08-25:
Yes, the power supply circuit will have problem when you are using external module.  For your references, I once tried to have 3.8V battery and will crash the radio when selecting on/off of an external multi-module.  However, charging the battery to 4.2V and no problem at all.

raphaelcoeffic, 2021-09-14:
@pfeerick status? should be closed?

pfeerick, 2021-09-15:
I can't reproduce this myself, but my NV14 does have an upgraded power supply circuit, so if it is indeed power related, it's certainly not something we can fix. I can turn my ELRS module on and off multiple times and it won't hiccup. I'm thinking this can probably be closed until we can point the finger conclusively at something the firmware is doing differently... but first...

@petervieth Can you try the latest nightly and let us know if there has been any change in behaviour?

petervieth, 2021-09-15:
Hey folks, I'm buying a house and just don't have time to fiddle with the
NV14 for a few weeks. I guess you can close this issue.  I'll look into
using an external battery instead of the two 18650s or lowering the power
on the Xfire; I had the NV14 shut off flying a whoop with the battery
indicator at a half, not reassuring!