EdgeTX/edgetx #613: [Bug] [NV14] ExpressLRS does not work (any more)

LimitlessGreen, 2021-08-21:
ExpressLRS on the NV14 is broken. I tested 83413a22425763b16139b4352604176534811fca and 478e345a32ab299402bf891096b3b6b2957af862, neither of them worked. The last working version I tested before was ff5e15cce72aa255f117328788fca9a5fe04f75e. So somewhere between ff5e15cce72aa255f117328788fca9a5fe04f75e and 83413a22425763b16139b4352604176534811fca the breaking changes happened.

pfeerick, 2021-08-21:
Does the radio lock up as well? If so, this is a duplicate of #610.
Otherwise, it *could* be related to https://github.com/EdgeTX/edgetx/pull/593.

Since you have a testable CRSF/ELRS setup (my NV14 is out in the car, and I've yet to coax my ELRS module on it due to my module not having a recess)... can you try the firmware artifact from the commit immediately prior? That will help narrow it down further. https://github.com/EdgeTX/edgetx/runs/3327957730

And if you are really bored, this is the next build, about three commits later... https://github.com/EdgeTX/edgetx/runs/3350354014

LimitlessGreen, 2021-08-21:
- I don't have the lock up issue from #610 
- run 3327957730 does work
- run 3350354014 does not work

pfeerick, 2021-08-22:
@richardclli What did you break? 😆 ^^^

LimitlessGreen, 2021-08-22:
No problem :)
I can't even connect to the module via LUA script too. Maybe a baud rate thing?

raphaelcoeffic, 2021-08-22:
> Thanks for testing that out! :)
> 
> @richardclli What did you break? 😆 ^^^ (it seems like the #593 telemetry fix broke CRSF telemetry somehow)

I'll check right now. @3djc pointed me at another issue related to external MPM which needs to be checked as well, which seems related to signal polarity on PPM pin. These 2-3 issues could be related.

richardclli, 2021-08-22:
> 
> 
> > Thanks for testing that out! :)
> > @richardclli What did you break? 😆 ^^^ (it seems like the #593 telemetry fix broke CRSF telemetry somehow)
> 
Maybe, the point is INV or not need to be detected and the current way to do it is by what user selected.  For multimodule, it does not work before my fix.

What need to be selected for CRSF?  PPM?

LimitlessGreen, 2021-08-22:
> What need to be selected for CRSF? PPM?

CRSF needs CRSF to be selected :octocat:

pfeerick, 2021-08-22:
> Btw, why does run 3350354014 reference commit [67c47cf](https://github.com/EdgeTX/edgetx/commit/67c47cf1678fe4185227d052f63a118f126eacb0) while my remote control says it's commit [3b3d34b](https://github.com/EdgeTX/edgetx/commit/3b3d34be0689c14b8cbd2fd430a267d38e96a9b1)?

It's a 'quirk' of the build system - it's building as if it is about to be merged into the code, so that is the commit id of the merge if that were to happen. It has not led to any confusion _at all_ when trying to figure out which commit I'm running prior to the pull request being merged. 😆

LimitlessGreen, 2021-08-22:
You made me curious, and I took a quick look at your git workflow.
<details>
  <summary>Show image!</summary>
 
[image]
  
</details>

You should rebase before merging. It's a bit more clean and bulletproof (and makes things easier to undo). But other than that, I like the dev workflow here! :)

But yeah, it's going a bit off-topic...sorry :shipit:

pfeerick, 2021-08-22:
Meh, it happens... and yes, rebasing is good... and for PRs I do I do that every now and then if they are open for a while... but unless there is some overlap, it's not big deal either way... just means when you checkout a PR to try it, and stuff has been fixed in the mean time, the resulting firmware build can give feelings of deja vu! 😆

commit 9509569c (2021-08-22): [nv14] aligned telemetry signal polarity on other radios and fixed requested polarity for MPM

Resolves #590
Resolves #613
closed 2021-08-22 by raphaelcoeffic