EdgeTX/edgetx #620: [UI] [NV14] more UI stuff

LimitlessGreen, 2021-08-22:
[image]
- the **Trainer mode** button is very hard to not possible to reach at the end of the page (more space there?)
- the top bar is for small fingers (not sure if it's good or bad)
- the **CH8** field does not react

**General information**
 - EdgeTX version (2.5.0 nightly, 0929e001b26f6d48b1897fbde4ec21b8cd172acb)
 - Hardware (FlySky UFPV Nirvana NV14)

eshifri, 2021-08-23:
The same problem exists on TX16S as well (and, most likely, on all color screen TXs).
If you play for some time with trainer settings, the bottom line will go below the bottom of the screen.
I think something is wrong with line 513 on model_setup.cpp:
      getParent()->moveWindowsTop(top() + 1, adjustHeight());
when height adjustment is negative something goes wrong.

richardclli, 2021-08-23:
Yes, the top bar is quite small, but it is now using the same settings as TX16S did.  So I think we need to consider the screen DPI as well, so for NV14, as the screen is smaller when comparing to TX16S, we may need to use a set of larger icons, maybe.

pfeerick, 2021-08-23:
Overall, touch screen UI needs more work to allow for screen edges/bezels and "fat finger" ;)

Not being able to edit the CH8 (channel end field) is consistent with OTX so not a bug.

eshifri, 2021-08-23:
Please notice that if you play long enough with trainer options and then set it to OFF, the option to enable it will be outside the screen and unavailable. To get it back you have to go to another screen and return back.
Screen resizing is messed there.

eshifri, 2021-08-23:
I think the problem is caused by the call setInnerHeight(innerHeight + delta);
in the void Window::moveWindowsTop(coord_t y, coord_t delta)
line 504 in window.cpp
https://github.com/EdgeTX/libopenui/blob/master/src/window.cpp#:~:text=%7D-,void%20Window%3A%3AmoveWindowsTop(coord_t%20y%2C%20coord_t%20delta),%7D,-void%20Window%3A%3Ainvalidate

Why do we need to change the window inner size, ehn changing the position of the top?

Commenting this line out fixies the problem.
I'm reluctant to propose a PR, since I do not understand why this call was introduced.

Can somebody better familiar with libopneui take a look please?

The problem with difficulty of acces to the last line on the screen can be resolved by adding one more
grid.nextLine(); in model_setup.cpp line 1391:

// Trainer
  {
    new Subtitle(window, grid.getLineSlot(), STR_TRAINER);
    grid.nextLine();
    grid.addWindow(new TrainerModuleWindow(window, {0, grid.getWindowHeight(), LCD_W, 0}));
  }
// add grid.nextLine() here
  window->setInnerHeight(grid.getWindowHeight());
}

pfeerick, 2021-08-23:
It's not new either - it's been there for two years... 😆 ... just have a look at the blame for that line... points to this commit... https://github.com/EdgeTX/libopenui/commit/0a22b09c12ba2cad7ce79f5ff2f755120afcadee

eshifri, 2021-08-24:
I have to admit, that I do not understand the previous version as well. :-)
Why moving the top requires change to the inner size?

pfeerick, 2021-08-24:
Something to do with the height? If you move  the top, the bottom moves as well as height? My heads in layout la-la land, so that's the most you're gonna get from me atm. 😆

eshifri, 2021-08-24:
In order calculate the shift we call Window::adjustHeight() which already changes the window size.

closed 2021-08-31 by pfeerick