EdgeTX/edgetx #622: [NV14] add a way to interact with LUA scripts (expressLRS in particular)

rentacheese, 2021-08-22:
**Is your feature request related to a problem? Please describe.**
Yes, so interacting with the LUA script is impossible with the touch screen, and right now, all the trims do when press in are enter or back. 

**Describe the solution you'd like**
I want to use the menus in the LUA scripts by perhaps rebinding the trim switches to up and down. I don't use the trim switches at all anyways, so having this as an option to rebind the trim switches would be a great addition

raphaelcoeffic, 2021-08-23:
The idea has been floating around of having 2 modes for the trims, which could be switched by a long-press on the center button (which is useless for trims). That could be an easy work-around to allow for quick compat.

pfeerick, 2021-08-23:
Another thought might be simultaneous press of the two trims be the toggle between modes if that can be monitored for - as a long press does mean a long ENT would be lost on the NV14... I didn't think of that when I said that before...

raphaelcoeffic, 2021-08-23:
> Another thought might be (near) simultaneous press of the two trims be the toggle between modes if that can be monitored for - as a long press does mean a long ENT would be lost on the NV14... I didn't think of that when I said that before... Maybe also warrants a 'default state' toggle in the radio settings? So if you choose between trims or navigation as the power-on default?

Detecting "simultaneous" press could be prove to be tricky, I'd try to avoid it. If we want it simple, we could just as well declare trims to be navigation in standalone LUA scripts, whatever happens? Of course, this would not work with widgets in full-screen, but as it is fairly new, we can assume that these widgets can start directly with implementing touch support.

JimB40, 2021-08-23:
Hey definitely thing to implement. To my finding the best UX is combining touch with physical inputs.
Right one should be navigation plus ENTER on press while left can be binded to PG>/PG</MDL/SYS/TELE plys EXIT on press

Then switching between trim mode and nav mode.
I liked @pfeerick concept of simultaneous press although EXIT and ENTER might no be good combinantion :)

richardclli, 2021-08-31:
Old drivers will automatically changed to stick mode when it detects the context is LUA script and get back to trim mode when it is out of LUA context, this can be another idea to make it works.

raphaelcoeffic, 2021-08-31:
> Old drivers will automatically changed to stick mode when it detects the context is LUA script and get back to trim mode when it is out of LUA context, this can be another idea to make it works.

This would be ok for “stand alone scripts”, which are normally executed on the ground. Full screen widgets will not be able to use this, as you might really need to the trims while flying and having a full screen widget on.

JimB40, 2021-09-02:
> Old drivers will automatically changed to stick mode when it detects the context is LUA script and get back to trim mode when it is out of LUA context, this can be another idea to make it works.

@richardclli I find mix of Touch and buttons use most efficient so nawigating through ETX with joystick(s) would be a plus if we can accomplish this trims function change mode

closed 2021-11-09 by pfeerick