EdgeTX/edgetx #623: Toggle uninverted Sbus on the TX16S UART

CamGenius, 2021-08-23:
Not sure if this is a hardware limitation but if we could get a software toggle to switch to uninverted sbus on the UART for the TX16S that would be awesome. This would allow users to use radiomaster receivers and futaba receivers for wireless trainer without needing an external inverter.

raphaelcoeffic, 2021-08-23:
> Not sure if this is a hardware limitation but if we could get a software toggle to switch to uninverted sbus on the UART for the TX16S that would be awesome. This would allow users to use radiomaster receivers and futaba receivers for wireless trainer without needing an external inverter.

So this is on Trainer input? Indeed, there is no setting here.

CamGenius, 2021-08-23:
I imagine it would go on the serial port settings on the hardware page because this feature wouldn't be exclusive to just wireless trainer (that was my use case/example) but could be used for any external custom addons one might add.

gagarinlg, 2021-11-07:
I checked to processor manual. In the processor used in the TX16s the uart pins can not be inverted in hardware.
Are there enough resources available to implement a software uart to receive sbus data?

raphaelcoeffic, 2021-11-07:
> I checked to processor manual. In the processor used in the TX16s the uart pins can not be inverted in hardware. Are there enough resources available to implement a software uart to receive sbus data?

That amounts to the availability of a proper hw timer with a free channel on the needed port. Then it can be supported in a fairly effective way.

gagarinlg, 2021-11-07:
I should ask my questions more precisely. What I meant to ask was, if there is a hardware timer free, or if they are all used up in the TX16s.
Implementation of a fixed frequency uart is simple.

raphaelcoeffic, 2021-11-07:
My beloved data sheets, not so easy in a mobile phone 😇

rotorman, 2021-11-08:
TX16S AUX1 RX pin is wired to STM32F429BIT6 PB11 that can be attached to timer 2 channel 4.
TX16S AUX2 RX pin is wired to PG9 that has no hw-timer connectivity.