EdgeTX/edgetx #628: Tidy Up Translations

richardclli, 2021-08-24:
Translation codes are quite messy at this moment.

Will try to tidy up the translations definitions, source generations, etc.

**General information**
 - EdgeTX main nightly
 - Hardware all

ulfhedlund, 2022-12-14:
I just ran the lupdate command on my local main branch and there were a few warnings, four of them related to context UpdateCompanion that I tried to fix (but chances are not at all correctly). A developer needs to do that, not me.

lupdate ./companion -ts ./companion/src/translations/companion_sv.ts

./main/companion/src/updates/updatecompanion.cpp:36: tr() cannot be called without context
./main/companion/src/updates/updatecompanion.cpp:41: tr() cannot be called without context
./main/companion/src/updates/updatecompanion.cpp:46: tr() cannot be called without context
./main/companion/src/updates/updatecompanion.cpp:52: tr() cannot be called without context

richardclli, 2023-02-24:
Not relevant anymore, close it.

closed 2023-02-24 by richardclli