EdgeTX/edgetx #632: Compilation failing since yesterday's commit

stefaanbolle, 2021-08-24:
Since commit e9552933f9a1c66333b6204f6b5c17742e192c8b in main repo
Compilation is failing

`In file included from /home/admin/edgetx/edgetx_main/radio/src/gui/colorlcd/libopenui.h:29,
                 from /home/admin/edgetx/edgetx_main/radio/src/opentx.h:41,
                 from /home/admin/edgetx/edgetx_main/radio/src/targets/common/arm/stm32/cpu_id.cpp:21:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h: In constructor ‘StaticText::StaticText(Window*, const rect_t&, std::string, WindowFlags, LcdFlags)’:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h:33:28: error: ‘FIELD_FRAME_COLOR’ was not declared in this scope
   33 |         setBackgroundColor(FIELD_FRAME_COLOR);
      |                            ^~~~~~~~~~~~~~~~~
In file included from /home/admin/edgetx/edgetx_main/radio/src/gui/colorlcd/libopenui.h:29,
                 from /home/admin/edgetx/edgetx_main/radio/src/opentx.h:41,
                 from /home/admin/edgetx/edgetx_main/radio/src/targets/horus/usb_charger_driver.cpp:21:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h: In constructor ‘StaticText::StaticText(Window*, const rect_t&, std::string, WindowFlags, LcdFlags)’:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h:33:28: error: ‘FIELD_FRAME_COLOR’ was not declared in this scope
   33 |         setBackgroundColor(FIELD_FRAME_COLOR);
      |                            ^~~~~~~~~~~~~~~~~
In file included from /home/admin/edgetx/edgetx_main/radio/src/gui/colorlcd/libopenui.h:29,
                 from /home/admin/edgetx/edgetx_main/radio/src/opentx.h:41,
                 from /home/admin/edgetx/edgetx_main/radio/src/targets/horus/haptic_driver.cpp:21:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h: In constructor ‘StaticText::StaticText(Window*, const rect_t&, std::string, WindowFlags, LcdFlags)’:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h:33:28: error: ‘FIELD_FRAME_COLOR’ was not declared in this scope
   33 |         setBackgroundColor(FIELD_FRAME_COLOR);
      |                            ^~~~~~~~~~~~~~~~~
In file included from /home/admin/edgetx/edgetx_main/radio/src/gui/colorlcd/libopenui.h:29,
                 from /home/admin/edgetx/edgetx_main/radio/src/opentx.h:41,
                 from /home/admin/edgetx/edgetx_main/radio/src/targets/common/arm/stm32/rotary_encoder_driver.cpp:21:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h: In constructor ‘StaticText::StaticText(Window*, const rect_t&, std::string, WindowFlags, LcdFlags)’:
/home/admin/edgetx/edgetx_main/radio/src/thirdparty/libopenui/src/static.h:33:28: error: ‘FIELD_FRAME_COLOR’ was not declared in this scope
   33 |         setBackgroundColor(FIELD_FRAME_COLOR);
      |                            ^~~~~~~~~~~~~~~~~
make[3]: *** [radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/build.make:97: radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/__/horus/haptic_driver.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/build.make:123: radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/__/horus/usb_charger_driver.cpp.o] Error 1
make[3]: *** [radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/build.make:110: radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/__/common/arm/stm32/cpu_id.cpp.o] Error 1

make[2]: *** [CMakeFiles/Makefile2:1750: radio/src/targets/simu/CMakeFiles/opentx-tx16s-simulator.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1727: radio/src/targets/simu/CMakeFiles/libsimulator.dir/rule] Error 2
make: *** [Makefile:732: libsimulator] Error 2`

pfeerick, 2021-08-24:
Anything to do with `libopenui` almost automatically means you haven't updated the submodule yet... 

Run `git submodule update` in the repo root, and you should be fine.

stefaanbolle, 2021-08-24:
That did the magic, thanks!

closed 2021-08-24 by stefaanbolle