EdgeTX/edgetx #638: [2.5 UI] Bootloader and Startup screen background

JimB40, 2021-08-25:
So we already started this with @raphaelcoeffic but stalled waiting for 2.5 UI rework

So here is preview
1. Startup screen
[image]
2. Bootloader
[image]

Do here is how to do:
1. Tile (535 bytes)
[image]
2. Full Mask (48kB)
[image]
3. Quarter Mask (12kB)
[image]

So to reduce mem usage trick is to get Quarter mask duplicate 3 times, rotate and position in right place
Should be 100% symmetrical. If not I'll replace it
@pfeerick @richardclli feasible?

pfeerick, 2022-09-09:
This will definitely happen in 2.9, if only because we now want to better support different screen sizes! 😆

froqstar, 2023-04-14:
That mesh background feels a bit too much like 2000 era design to me. 
Isn't the trend towards more flat design? I would even prefer a completely black/grey background TBH. But that's just me.

pfeerick, 2023-04-14:
Radio trend seems to be more towards things like carbon fiber look and colour atm, but like everything it has its cycles... todays 2000 era "boring" is 2025's craze again.

pfeerick, 2023-06-21:
#2764 only resolves this in part, I don't think it touched the bootloader background, only the default splash screen background.

JimB40, 2023-06-21:
@pfeerick I dont insist on fancy background. We can go with simple black bkg. Just not that cheese :) 
Also this concept to display EdgeTX version and build under logo on splash screen has advantage. You see trouble shooting picture or video and you don't have to ask user what version he has. This should resolve many issues with wrong firmwares loaded

pfeerick, 2023-06-22:
> We can go with simple black bkg.

I'm actually starting to think looking that second image in the first post a flat / matt background for the bootloader might be better... the honeycomb actually makes it quite noisy - the current one at stays in the background more. 

> Also this concept to display EdgeTX version and build under logo on splash screen has advantage. You see trouble shooting picture or video and you don't have to ask user what version he has. This should resolve many issues with wrong firmwares loaded

That will only apply to the default splash screen, so will invariably still need to ask the majority of users to go to the SYS -> Version page, but yes, it should help a bit.

PR #2764 (2023-04-14): Work out a dynamic splash screen that fits all screen sizes.

Fixes #638

Summary of changes:
PR #4139 (2023-10-03): feat(color): Remove splash screen and boot loader background, add version info
Also display version and codename on splash screen.

Fixes #4134, resolves #638
closed 2023-10-03 by pfeerick